ESPresense/ESPresense #1119: Static IP during setup

naudor, 2023-11-12:
### Is your feature request related to a problem? Please describe.

I would like set static IP for ESPresence stations. It's more efficient and allow to have more control

### Describe the solution you'd like

I think would be more easy just after install firmware, when you set the Wifi name and password, but obviously the better solutions would be the possibility to edit in captive web

DTTerastar, 2023-11-13:
Dup of #709

DTTerastar, 2023-11-13:
It wasn't popular enough for someone to try and implement. I fear this one would be similar.  But you never know...

CV8R, 2024-01-06:
Commenting here because static IP assignment would benefit my use case.  One of my devices, a wired WT32-ETH01 is on a network at the far end of a 250 meter wireless link that is serving my front gate. The system is powered by a solar system.  Granted it is rare, but occasionally the system is powered down (for maintenance, solar outage, etc).  The DHCP server is at the other end of that link and the wireless link takes a while to come up at which point devices often give up on DHCP and default.  As such, I prefer to set all devices at the far end to Static IP.

lucacri, 2024-02-04:
This, and also wifi settings at creation of the image would be fantastic. I am about to embark in setting up ~10 espresense and it'll be a bit laborious (connect to AP, change settings, switch back the wifi, etc)

SkipperDo, 2024-05-25:
I would like the option for static IP addresses of the nodes, too...
I have a DHCP address pool for mobile Devices & guests who are allowed to log into my WiFi.. (only a range of 20 addresses)
I also have a configured "Node-Address-Pool" who is outside the DHCP address pool.... 

Its a way to clean up the Network...

So i would prefer a way to configure the IP of any ESPresence Node..

rolfvreijdenberger, 2024-06-10:
+1 on static ip. I am using 3 esp32 based projects for home assistant, wled, esphome for generic stuff and espresense.
Of these 3 wled and esphome offer static ip setting. It would be great if espresense offered this.

mcsdodo, 2024-06-17:
+1 for this feature

DonPosei, 2024-07-06:
+1 for this feature

FARKIr, 2024-11-07:
I would also greatly appreciate the option to set a static IP for ESPresense nodes. This setup would provide more stability and control over network connections, especially in situations where DHCP server access is limited or for certain applications requiring a reliable network address. I believe this feature would be beneficial for many users who utilize ESPresense for home automation.

swampdogmash, 2025-01-24:
+1 on this also please. Whilst I can set a static DHCP address, I have configured all my IOT devices with static IPs, so that if there is no DHCP server, things still work. I may not have internet (router down) but that doesn't mean that for example, homeassistant and the ESPresense devices aren't up and working.

Andy

SupremeSports, 2025-02-04:
+1, I like to have my IOT devices in specific IP range for easy access and decluttering

DTTerastar, 2026-11-01:
Closing this as not planned. Use a DHCP reservation on your router: you get the same fixed address, you can put reserved addresses in their own range, and you can manage them all in one place. Thanks to everyone who weighed in.

closed 2026-11-01 by DTTerastar