ESPresense/ESPresense #1128: MQTT connectivity with mosquitto broker on HA (ESP32)

wongkaiyang11, 2023-11-21:
### Describe the bug

So i have installed espresence 3.3.1 on my esp32 and configured everything on Home assistant to detect my devices, but the issue lies in the fact that my esp32 becomes disconnected when i dont open the port i used to install espresence and go to the part where i can see the choices. "Change wifi, visit device etc)

This is affecting me as i want to set it up by itself not hooked constantly on my laptop, please any assistance?

### Version

3.3.1

### Active scan enabled

NA

### Include filter

NA

### Exclude filter

NA

### Query filter

NA

### Firmware flavor

NA

### Device (be specific)

NA

### Logs

```text
Disconnected from MQTT; reason 0
0 Reconnect timer
```

wongkaiyang11, 2023-11-21:
i have to add that everytime i refresh the page it asks me to install espresense again, i have to then go to logs and wait till there is text shown on the terminal and press "back" for it to let me visit device,change wifi etc.

ediciuStsaF, 2024-01-15:
I have the problem that  the ESP32-C3 Supermini gets programmed and recognized one Homeassistant but i can´t get the console or log to work. so i can´t use it at all. Can´t see connected devices to determine the ones i need.

Some news about this or an older version known that works?

closed 2024-09-09 by stale[bot]