ESPresense/ESPresense #1129: /ui/#/settings page does not show up

labob246, 2023-11-22:
### Describe the bug

In version 3.3.1 the /ui/#/settings page does not show (see screenshot).  I have 9 espresense devices and some settings pages show after upgrading some do not.  It does not seem to be the type of device (ESP32 D1 mini or ESP32 Dev board) specific.  All the devices have the same info in them (except for Room) and the default configuration except for filter.  The devices page and fingerprints page work fine.  I have tried 4 browsers, some with fresh installs and it does not seem to help.  Since most options are available in Home Assistant, I don't care, but I do need the change the filters option every now and then and cannot see it.

### Version

3.3.1

### Screenshots

[image]

nybdog, 2023-11-24:
having the same problem.

SheckieJR, 2023-11-26:
Here as well.  The settings do appear, but then disappear for me.  That led me to believe it was browser or extension related, so I also did that testing, to no avail.
[image]

sriprasadgit, 2023-12-06:
havign the same issue, in my one of device

jfiggins, 2023-12-08:
[image]

Same issue here, only on one of my devices though.

ecto1a2003, 2023-12-18:
All the same here

markheitz, 2023-12-18:
First time with espresence.. just installed and having the same errors also.

Schlafenein, 2023-12-25:
same issue here, only one of three have this issue. all v3.3.1

ecto1a2003, 2023-12-25:
> full erase and reflash seems to fix it for me currently

dacorn, 2023-12-30:
I have the same issue on 5 of 8 units. All have the same version installed. 3.3.1 Do we have an ETA to get this fixed as I need to add more devices to track. Or is there another way to add resolving keys? Thanks guys!!

dxmnkd316, 2024-01-03:
Looks like it has to do with a bad json format.  I accidentally pasted a string with either a tab or extra space into the device filter and it gets hung up on loading the "/extras" string.  

Not sure how to fix it.  That's just what the issue is for me.  

Going to try to reflash, which I'm confident will work.  Just a PITA.

roofuskit, 2024-01-20:
Tested in Firefox and Chrome. Same issue.

labob246, 2024-03-14:
I have updated all 9 of my epresense sensors to 3.3.2 and the problem seems to have disappeared.

closed 2024-09-28 by stale[bot]

arbiengine, 2025-04-18:
I dont think this was resolved, I am still not seeing the settings page after flashing with platform io to my esp32s3 board