ESPresense/ESPresense #1131: Only works when powered though computer USB

KevinBalkoski, 2023-11-27:
### Describe the bug

I have setup 2 ESP32's with ESPresense. When first getting everything set up, I  was connected to my PC and everything appeared to work. I was able to connect via MQTT and had everything looking good on my Home Assistant.

I then moved them to the rooms they need to belong to using USB power supplies. Then, I could not find them on my network. To troubleshoot, I brought them back to the office and plugged them into the outlet (but they were physically in the same location where I set them up).... and they didn't show up. I then plugged it back into my computer, and suddenly they were back up. I've tried multiple cables and USB power supplies. If I have them plugged into my computer, they work 75% of the time but 0% of the time from any power supply.

Are my ESP32's just duds or is there something else going on?

ESP32's: [AITRIP ESP32S ESP32 ESP-WROOM-32 Development Board 2.4GHz Dual-Core WiFi +Bluetooth 2 Function Microcontroller for Arduino ](https://www.amazon.com/ESP-WROOM-32-Development-Microcontroller-Integrated-Compatible/dp/B0B11N1X8R?crid=2R7B6EY31OF2V&keywords=LOLIN32%2BESP32&qid=1655589767&sprefix=lolin32%2Besp32%2Caps%2C64&sr=8-5&linkCode=sl1&tag=espresense-20&linkId=cc9dcfa47f47f6fe47e4c697535f814c&language=en_US&ref_=as_li_ss_tl&th=1)
Power supplies: [5V, 2.4A Amazon Basics 12W One-Port USB-A Wall Charger (2.4A)](https://www.amazon.com/gp/product/B0773J79KC/ref=ppx_yo_dt_b_asin_title_o01_s02?ie=UTF8&th=1)
USB Cables: [Amazon Basics](https://www.amazon.com/gp/product/B018RAVRNS/ref=ppx_yo_dt_b_asin_image_o01_s01?ie=UTF8&th=1)
My PC: desktop PC running Pop OS

I have tried different supplies and cables. The cable that seems to work most consistently is a very short ~6in cable.

The logs attached are from just one that was working (although it did seem to randomly restart).

### Version

3.3.1

### Firmware flavor

Standard

### Device (be specific)

AITRIP ESP32S ESP32 ESP-WROOM-32 Development Board 

### Logs

```text
1 New    | c74189dc60fb | c74189dc60fb                                              -97dBm 
1 New    | 647980142a0f | 647980142a0f                                              -93dBm 
1 New    | 5f49b6240178 | apple:1005:9-12                                           -94dBm 131852fc7f
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 215796
Setup network
Connecting to WiFi SSID 'SSID-6.0'............................................................................................................................................................................................192.168.1.30
Firmware:     esp32
Version:      v3.3.1
WiFi BSSID:   3C:7C:3F:CF:9F:30 (channel=1 rssi=-30)
IP address:   192.168.1.30
DNS address:  192.168.1.4
Hostname:     espresense-bedroom
Room:         Bedroom
Mqtt server:  192.168.1.44:1883
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   980d6f58c04b f4fefb7cc3eb
Count Ids:    
Post-Setup Free Mem: 147060

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 New    | abc213 | apple:1111:1-11                                          -89dBm 06243a1458
...

heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_realloc was called but failed to allocate 96 bytes with 0x1800 capabilities. 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093d24  PS      : 0x00060f30  A0      : 0x800e47ef  A1      : 0x3ffe94d0  
A2      : 0xd4f540e7  A3      : 0xd4f540e3  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800e3731  A9      : 0x3ffe94a0  
A10     : 0x3ffe95d0  A11     : 0x00000000  A12     : 0x3ffe9440  A13     : 0x401cbb18  
A14     : 0x00000004  A15     : 0x3ffe93d1  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xd4f540e7  LBEG    : 0x400923d1  LEND    : 0x400923f3  LCOUNT  : 0xffffffff  

Backtrace: 0x40093d21:0x3ffe94d0 0x400e47ec:0x3ffe94e0 0x4010e61e:0x3ffe9500 0x400fe931:0x3ffe9660 0x400fb165:0x3ffe96d0 0x400f43e1:0x3ffe9740 0x400f403d:0x3ffe9780 0x400f4435:0x3ffe97a0 0x400fa4e6:0x3ffe97c0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 215796
Setup network
Connecting to WiFi SSID 'SSID-6.0'...............................................................................................................................................................................................................192.168.1.30
Firmware:     esp32
Version:      v3.3.1
WiFi BSSID:   3C:7C:3F:CF:9F:30 (channel=1 rssi=-29)
IP address:   192.168.1.30
DNS address:  192.168.1.4
Hostname:     espresense-bedroom
Room:         Bedroom
Mqtt server:  192.168.1.44:1883
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   980d6f58c04b f4fefb7cc3eb
Count Ids:    
Post-Setup Free Mem: 147196

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 New    | abc213 | apple:1111:1-11                                          -89dBm 06243a1458
1 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-208-13516 to node:bedroom
...
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-232-7902 to {"id":"node:office","name":"Office"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-208-13516 to {"id":"node:bedroom","name":"Bedroom"}
```

wongkaiyang11, 2024-01-19:
yup having the exact issue.

boredomwontgetus, 2024-01-31:
yeah exactly the same for me. I have 10 (!) ESP32 flashed and working as long as they are connected to a serial port. I tried 8 different cables and 5 different power supplies. Also powering the devices on the 5v and GND pin does not help. The only way to get it connecting to WiFi is powering it from any (multiple PCs tried) USB port.

Zitus, 2024-02-03:
...same issue discovered! What is going on? My esp32 code won't run on 3.3v, 5v, or powerbank USB! only works when powered from the computer USB

boredomwontgetus, 2024-02-03:
I did a bit of trial and error and it turned out that i can make it power on from any 5V source when either holding down "BOOT" while plugging it in or by pulling down GPIO0 to ground with a 10k resistor. I have never seen this kind of issue on any nodemcu but for the ones i have (azdelivery) its the only way to get them going.

angristan, 2024-02-03:
I also had this issue on my azdelivery one. I bought three from diymore that work fine on any power supply. Might be anecdotal evidence though 🤔

Zitus, 2024-02-03:
On SeedXiaoC3 and LilyGo TQT-Pro , it happens, when the code is one that tries to scan for BLe devices. Doesn't happen when I upload other code so this might be code related. No pressing of any button helps with 5V.

borodaaaaa, 2024-02-08:
The same issue here. It does not work on usb adopters unless you connect it to a pc usb. I see ping drops a lot or non at all if on wall usb plug. web gui does not load at all...

ivoras, 2024-08-17:
bump - same issue

borodaaaaa, 2024-08-17:
Experiments with power supply and cables make good results. It is surly must be more stable than that!  You can also try to the play with disabling captive portal this will at least not let it go into it if it has issues with boot

Peacem4kr, 2024-08-30:
i feel i have the same issue, but i also feel that it does work with a mobile charger if  you would remove the sensor from the ESP, then connect the power, then connect sensors.
But this is not realy handy for people that solder the connections, of even just moving the device to a different location.

ThomasDuffner, 2024-11-04:
I found this and it works for me:
replace build flags
-D ARDUINO_USB_MODE=1
-D ARDUINO_USB_CDC_ON_BOOT=1
with
-D ARDUINO_USB_MODE=0
-D ARDUINO_USB_CDC_ON_BOOT=0

https://github.com/espressif/arduino-esp32/issues/7554

tested with:
Waveshare ESP32-S3 Mini

ricardoacosta290, 2024-11-16:
I had the exact same issue with an AZDelivery ESP32 while using ESPHome to control a BLE device. When connected to the PC via USB, it worked fine, but when using different cables or chargers, it would boot up, but BLE wouldn’t function correctly. After trying several options, the problem was resolved by disabling the USB port for data (you can still power it via USB) with USB_MODE=0.

`esphome:
  name: esp32-bluetooth-proxy
  friendly_name: Bluetooth Proxy
  platformio_options:
    build_flags: 
      -D USB_MODE=0
      -D USB_CDC_ON_BOOT=0`

Knobee, 2024-12-11:
Same issue as #998 I believe.

Knobee, 2024-12-24:
https://github.com/espressif/arduino-esp32/issues/7554#issuecomment-1969614096
 
(?)

Knobee, 2024-12-24:
I have been able to correct this problem with a one line change in the platformio.ini file:

The line:

`platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.07.00/platform-espressif32.zip
should be replaced with:
`platform = https://github.com/platformio/platform-espressif32.git`

This brings the built image up to "pre-release" on the Expressif32 code, but it solves the problem.

For those that are interested, it appears that the old Expressif code expects there to be a serial console attached when the chips are powered by the USB port and when you just have a power adapter attached, the board crashes (or halts) after just a few seconds.  This is NOT the fault of "some cheap knockoff" boards, this is a natural progression of the hardware that was not backwards compatible with the existing software.

I have no idea why ESPsense is locked to that (2023.07.00) version of the code, but the change to use the more current has not caused me any downsides at the moment.

Would it be possible for the maintainers of the project to make this one-line change so that everyone on this thread can get back to using the web installer?  :-)  I've learned more about VisualStudio Code and PlatformIO than I really ever wanted to... but I feel like I stayed in a Holiday Inn last night.

Now, I guess I should figure out how to do the update and submit it in the right way.... maybe someone will see this before I get it figured out.  :-) :-) :-)

fetalp, 2025-01-10:
Same issue as well, only works when connected to my computer's usb ports. Tried many different cables from USB to USB C  and charging block, even used USB C to USB C no change plug it in directly to PC no issue again.

Tried to figure out how to update the platformio.ini but not luck as of right now PlatformIO isn't an extension in visual code as it's been removed at this time.

jkok-ops, 2025-01-20:
Having the same issues, cant really use my ESPs for espresense like that. Could need some advise how to flash it with changed parameters or an older version. I have some IT Knowledge but im a noob with ESPs....
Help would be welcomed.

QNTLO, 2025-01-30:
Same issue here. Just bought a esp32-wroom and it is behaving exactly like described in this thread. Bought it for espresense, but it is useless in this state. Hope the web flasher will solve this, since I have not a clue on how to do it otherwise.

QNTLO, 2025-01-31:
> I found this and it works for me: replace build flags USB_MODE=1 USB_CDC_ON_BOOT=1 with USB_MODE=USB-OTG (TinyUSB) USB_CDC_ON_BOOT=0
> 
> [espressif/arduino-esp32#7554](https://github.com/espressif/arduino-esp32/issues/7554) Arduino IDE 2.3.3 tested with: Waveshare ESP32-S3 Mini

I can confirm this worked for me!
ESP32-WROOM

aljazjelen, 2025-08-09:
hey! any news on this topic?

ejs553, 2026-01-06:
Running into this same issue on v3.3.5. Any updates  on native fixes in the last year?

closed 2026-09-04 by stale[bot]