ESPresense/ESPresense #1136: Add support for WESP32 Rev7 or later

compuguru, 2023-11-29:
### Is your feature request related to a problem? Please describe.

The current configuration for the WESP32 ([link](https://github.com/ESPresense/ESPresense/blob/master/lib/network/Network.h#L84))  uses the `ETH_PHY_LAN8720` for the ethernet module.  However, according to the [docs for WESP32](https://wesp32.com/files/wESP32-Product-Brief.pdf), the ethernet module changed in Rev7 of the board to RTL8201.  So those boards should be using `ETH_PHY_RTL8201`

### Describe the solution you'd like

Add an option to select a specific option for Rev7 of the board

### Additional context

I can create a PR with the change if you'd like.  I'm assuming the new entry would have to go at the end so we don't change the index of the existing configured items.

CV8R, 2023-12-14:
+1 WESP v7 are fantastic high quality boards.

CV8R, 2024-03-14:
I have implemented the Rev 7 board in my fork and is working reliably for >48hrs.  I added the changes to have the type appear in the Ethernet Type drop-down and also added a board type.  The board type is added to define the default pin types which conflict with the defaults.  There is a clear warning from the vendor that not configuring the pins correctly can result in damage.

Question - the placeholder values in the GUI still say the old values and I could not see a way to override those for the defined pins.  I had confirmed via the log that the correct pins are used though.

Shall I create a pull request for the changes?

Incidentally, the risk is higher with the earlier board revisions, so the board type should probably be implemented there too but I do not have one of those to test with.