ESPresense/ESPresense #1142: Integrating with Home Assistant  - devices aren't recognized

theDontKnowGuy, 2023-11-30:
According to documentation, devices need to be added to the sensor configuration as such:
- platform: mqtt_room
  name: "My iPhone"
  device_id: "my_iphone"
  state_topic: "espresense/devices/my_iphone"
  timeout: 5
  away_timeout: 180

assuming I have a live mqtt topic named: espresense/devices/device_id
I see this topic get updated every few seconds, and has two son topics with the room it was recognized at.

Fully restarted HA after saving config. 
No devices appear.
Saw other people complain about this, some suggested to change the topic to espresence/rooms, not sure how would this solve the problem. 

Anything I am missing?

DTTerastar, 2023-12-01:
That looks good to me.  Have you searched your entities?  It doesn't get attached to a device.

theDontKnowGuy, 2023-12-01:
Sure, I only got a dead entity:
7 hours ago is when I last restarted HA.

[image]
[image]

theDontKnowGuy, 2023-12-01:
Thanks @DTTerastar but they are very well connected, all espresence , and I have six of them, are reporting and are alive via their mqtt communication. There is something off with the configuration of the topics. 

[image]

gregcohen, 2023-12-02:
I'm having this same issue. Only one of my devices is updating as it should in HA, but none of the others are. However, they are all updating in MQTT explorer. I've tried re-ordering the sensors, but it seems HA is only picking up one device--and it's the same device every time.

malballuk1, 2023-12-05:
I have just started using this and agree with gregcohen, I have two apple watches configured and I can see distance changes in mqtt explorer and in espresense figerprints but my sensors in homeassistant show not_home all the time

DTTerastar, 2023-12-14:
https://github.com/ESPresense/ESPresense/issues/1091

Mastiffen, 2023-12-14:
For me it was something else, but I have no idea what. I had restarted Home Assistant and the ESPresense devices several times (yes, I had set it to the new template, I found that first), but gave up and went to bed. Next morning it worked on both of my locations (cabin and home), and I have no idea why. It's been workign since.

malballuk1, 2023-12-14:
Got all mine working by following the instructions on calibration. I put the watch about 1 metre away from the sensor and watched the messages in mqtt explorer, I kept adjusting the rsi based on what I was seeing, also adjusting the absorption factor until I got close to 1 metre

DTTerastar, 2023-12-14:
Great, glad to hear you figured it out

theDontKnowGuy, 2023-12-14:
At least for me, this wasn't the case.

What did solve my issue of devices not being recognized in HA was (after a v*e*r*y long trials and errors):
I had my system up and running; everything worked fine.
Then I stopped getting HA events for people coming and going, although I saw their phone flipping in the MQTT explorer.

**What happened was that I installed HA companion app on their (family member) devices.**
**The result was a naming collision:** 

1. Not only their tracking device (HA -> Setting -> People -> person -> Track device became their iPhone (change it but this will not solve the issue)
**2. The name collides: My phone is John's iPhone. this is the same name the companion app sends, and the same name I had captured by the ESPresense enrolls. So from now on, HA prefers the app, just you don't know that. Even if the user removes the app, the tracking isn't coming back.**

**Solution:**
1. Going to MQTT, change espresense->settings->irk:xxxx->config topic json value id and name from John's iPhone and johns_iphone to something else such as John's iPhone ESP and johns_iphone_esp
2. restart all ESPresense nodes. ye, it is fun.
3. change your configuration yaml accordingly. I am using this in the sensor.yaml file like this:
- platform: mqtt_room
  name: "Johns iPhone ESP"
  device_id: "johns_iphone_espresence"
  state_topic: "espresense/devices/johns_iphone_espresence"
  timeout: 5
  away_timeout: 180
4. restart HA completely (not just the useless reload whatever)
5. remember to change the user's tracking device and automation for the new naming convention.
6. Vualla

Let me know if this solved your case too.

dogasantos, 2024-01-15:
Same problem here.
Tracked devices show as not_home.
Sensor shows distance:null with no updates. Mqtt receives messages just fine. Configuration.yaml looks good (exactly like the documentation). Still, no updates on HA for the state of my devices.
Tried to change to state_topic to espresense/rooms (checking the box as well) but also doesnt work. 
Logs of mqtt shows all data correctly (rssi and everything being updated just fine). Still, not_home.

not sure whats going on here

PedroLageTavares, 2024-01-16:
@dogasantos , I was also having that issue and the way I have fixed (still confirming) was to use the device_id as seen in MQTT instead of IRK. Here's my current config:

```
sensor:
  - platform: mqtt_room
    device_id: "iphone_pedro"
    name: 'Pedro Iphone15Plus'
    state_topic: 'espresense/devices/iphone_pedro'
    timeout: 10
    away_timeout: 120

```

dogasantos, 2024-01-16:
Hi there!I did that. Still not working- Douglas

dogasantos, 2024-01-16:
Hi @PedroLageTavares !
Thanks for the tip.

Actually, I already have this setup.
Here is my config:

```
  - platform: mqtt_room
    device_id: "irk:7xxxxxxxxxxxxxxxxxxxxx27"
    name: "Watch Douglas"
    state_topic: "espresense/devices/watch_douglas"
    timeout: 10
    away_timeout: 120

```
It matches the device name provided during the enrollment process: 

[image]

and the mqtt topic in use:

[image]

The HA logs shows the messages being received (mqtt again) just fine.

And yet:

[image]

Distance: null.

No entities created for that sensor. Sensor with 0 information.

My 4 ESPresense stations have this config:

```
  - platform: mqtt
    name: ESPresense-Escritorio
    state_topic: espresense/rooms/escritorio/status
    json_attributes_topic: espresense/rooms/escritorio/telemetry
    payload_on: online
    payload_off: offline
    device_class: connectivity
```

The config is compliant with the current documentation. Yet, still no go for any device I've tried to track.

Any help here would be very appreciated. I bought raspberry pi and those devices with this project in mind and a week later it still doesn't work.

Best,
Douglas

Electronlibre2012, 2024-05-14:
Hello all,

same boat here...all node dont integrate in HA...and understand why....

I HAVE TYPING in the "Home Assistant discovery topic prefix: " in each node : "espresence".

BUT MQTT in HA is AUTODISCOVERY ONLY if you have "homeassistant" in the prefix setting...

So i put "homeassistant/espresence" in each node, and all working as expect....

jaburges, 2024-06-19:
Hi there, 
So something is not working correctly. I too have MQTT messages coming in with the correct new set up:

espresense/devices/watch/ROOMNAME

I noticed that if I changed the discovery topic to "ABC"
Then the ESPresense base stations MQTT values changes to ABC/espresense/.....
BUT the devices didn't.

So it looks like the ESP base stations are configured with homeassistant discoverable settings BUT the devices are not.
A regular MQTT sensor is possible using the state topic by parsing the values from the topic `espresense/devices/[deviceid]`

So is the issue with the mqtt_room component?

jaburges, 2024-06-19:
ok the MQTT_ROOM documentation lists the devices as the correct set up, but then below under usage mentions rooms.
https://www.home-assistant.io/integrations/mqtt_room/#usage which isnt the correct usage

cjramseyer, 2025-01-04:
I have also had this issue with the distance being reported as null.
Finding this issue, along with the comment linked from the forum, the device ID and state topic MUST match exactly as it shows in MQTT Explorer.

For example, here is my sensor entry that WORKS!
- platform: mqtt_room
  name: "ESPresense CJs Phone"
  device_id: phone:cjs-phone
  state_topic: "espresense/devices/phone:cjs-phone"
  timeout: 10
  away_timeout: 180

Notice the device ID AND state_topic.  the phone:cjs-phone is exactly how it shows up in MQTT Explorer.
With this, the distance shows up correctly when viewing this entity via developer tools, correctly shows when my phone is in a given room or not.