ESPresense/ESPresense #1159: Ability to set I2C sensors reporting interval (to less frequent than every second)

FvanEgmond, 2023-12-10:
### Is your feature request related to a problem? Please describe.

I have SGP30 Air Quality sensors enabled, which report values every second to MQTT. Values are not needed that frequent though, and it can possibly lead to too much load on the MQTT server when I keep adding such sensors to my setup. 

### Describe the solution you'd like

Ability to add control the reporting to MQTT interval (in seconds) for i2C sensors - or at least for the SGP30 Air quality Sensor. 

### Describe alternatives you've considered

I'm using Home Assistant to show the values. Disabling the recorder for the SGP30 values helps control the amount of data stored (and using a second trigger sensor to store values less frequently). but a better option is to control the amount of data to be send to MQTT, so all this manual setup doesn't need happen either.     

### Additional context

Under ESPpresense 'Filtering', you have the 'Skip reporting if message age is less then...' option already, which is a great way to control the MQTT message flow for distance reporting. I'm looking for an equal option for i2C sensors. :-)

arne182, 2023-12-13:
I am wanting to add 6 more to my home and I have just made a PR to reduce the reporting to every 5 seconds https://github.com/ESPresense/ESPresense/pull/1163 an option from the UI might be nice or maybe even an averaging as the readings fluctuate a lot.

FvanEgmond, 2024-01-21:
In version 3.3.2, the interval has been increased to once every 60 seconds. That's a sufficient enhancement for this topic for now.

closed 2024-01-21 by FvanEgmond