ESPresense/ESPresense #1180: unprotected ssid being brocast for espresense devices

turboc1208, 2024-01-03:
### Describe the bug

When I look at my available wireless networks, I am seeing "open" ssid's associated with each of my espresense devices.  I am running version 3.3.1.  
1. Are these needed can they be turned off
2. can these be secured

### Version

V3.3.1

Coder84619, 2024-01-05:
When you are configuring ESPresence you can set the number of seconds to wait for WiFi before the captive portal appears, and seconds to wait in captive portal before rebooting. I set mine to 60 and 300, respectively. After properly configuring WiFi, I no longer see the captive portal SSID persisting.

turboc1208, 2024-01-05:
I found it. I had a firewall rule out of order causing it them to be blocked. Once I fixed that and reset them, they came back online just fine. ThanksSent from my iPhone

closed 2024-07-18 by stale[bot]