ESPresense/ESPresense #1181: Enrolling iPhone 15 Pro doesn't work

Coder84619, 2024-01-05:
### Describe the bug

I have an iPhone 15 Pro on iOS 17.2.1, and running ESPresence 3.3.1. I click Enroll in ESPresence, and I only get a 'regular' Bluetooth pairing prompt in iOS, not a "secure" prompt. After I tap Pair in iOS, I do NOT see ANY resulting device listed in ESPresence. 

### Version

3.3.1

### Device (be specific)

iPhone 15 Pro, iOS 17.2.1

adolfonovo, 2024-01-09:
Same here . Following

meetbryce, 2024-01-11:
From my iPhone it looks like the pairing is successful but ESPresense doesn't seem to recognise or acknowledge it in any way. The modal stays up until it times out.

[image]

adolfonovo, 2024-01-11:
> From my iPhone it looks like the pairing is successful but ESPresense doesn't seem to recognise or acknowledge it in any way. The modal stays up until it times out.
> 
> ![Screenshot 2024-01-11 at 13 37 05](https://private-user-images.githubusercontent.com/1845143/296017066-a74ece97-512f-4292-a67e-2d247b6ca4c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5OTg4NTQsIm5iZiI6MTcwNDk5ODU1NCwicGF0aCI6Ii8xODQ1MTQzLzI5NjAxNzA2Ni1hNzRlY2U5Ny01MTJmLTQyOTItYTY3ZS0yZDI0N2I2Y2E0YzkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMTFUMTg0MjM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmY1MDI5OGE3ZjI0ZGM1MDEyM2UwN2I3MWU1ZDI4YTcxMjcyOGIzZjgyZjcyZDM5YWQzZTYyM2M3ZWJhZGYyMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.uqLeG7z9Lrthw3fLalK6qNYXJI7NAgg-hsYtW8InHUE)

Yes! Same behavior. I also tried with the Apple Watch with out success

meetbryce, 2024-01-11:
@adolfonovo I was able to get it working by following the instructions [here](https://espresense.com/beacons/apple) under `Lookup Method (requires a Mac)`

I figured out which one it was based on Date Modified matching the date I got my phone.

[image]

Then I had to set up the sensor definition as shown below based on what I found [here](https://github.com/ESPresense/ESPresense/issues/1031#issuecomment-1780541169):

```yaml
- platform: mqtt_room
    name: "Bryce iPhone BLE"
    device_id: "irk:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    state_topic: "espresense/devices/irk:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
    timeout: 5
    away_timeout: 180
```

meetbryce, 2024-01-11:
Good to know! That's going to be become a problem for me in a few hours then when I try to add a 2nd device with a separate iCloud =(

adolfonovo, 2024-01-12:
Thans for the info…. And what do you think could be the reason? iOS 17? ESPresense bug?
I guess the first one because i tried the enroll feature with an android device and works perfectly

Coder84619, 2024-01-12:
@adolfonovo But I don't think Android phones use the randomized BT MACs with IRKs? I have no clue where the bug is, but it's clearly broken.

meetbryce, 2024-01-13:
I wasn't able to connect an iPhone 14 Pro though the UI, so I suspect it's iOS 17 related.

adolfonovo, 2024-01-13:
> I wasn't able to connect an iPhone 14 Pro though the UI, so I suspect it's iOS 17 related.

Yes, almost certainly. Let's hope the developers can overcome that block.

meetbryce, 2024-01-13:
The workaround I shared above worked fine though, even with a different iCloud account. They just had to use their mac to get the `irk` for me.

Coder84619, 2024-01-18:
@meetbryce But I don't see a way to configure a device name? So the enrolled devices list blank, even if the IRKs are used to decode the MACs.

dauer1205, 2024-01-24:
Same issue with iphone 14 pro on ios 17.3

Cyberbeni, 2024-02-27:
I've connected to ESPresense with the [nRF Connect app](https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403) that was mentioned in the [original PR](https://github.com/ESPresense/ESPresense/pull/553) and it seems that it properly paired with my phone. (Although I had to also restart ESPresense because it went back to using the "apple:100x:..." id after like half a minute the first time)

Cyberbeni, 2024-02-28:
Did anyone try this method for getting the irk? https://community.home-assistant.io/t/guide-getting-the-irk-for-iphone-and-apple-watch-without-macos-maybe-also-other-devices/623694

The code seems a bit different, so it might work.

jtowe1, 2024-03-01:
i'm having the same problem, the countdown stops.  It looks like the websocket is disconnecting

[image]

Cyberbeni, 2024-03-04:
> Did anyone try this method for getting the irk? https://community.home-assistant.io/t/guide-getting-the-irk-for-iphone-and-apple-watch-without-macos-maybe-also-other-devices/623694
> 
> The code seems a bit different, so it might work.

I had time to try this out and it worked 2 out of 2 times with iOS 17.3.1 devices.

FYI @DTTerastar

DrShivang, 2024-03-27:
+1 Iphone 13 Pro with IOS 16.5.1, the enrol popup times out with nothing in messages too

bzumik1, 2024-04-10:
Same issue here as well.

luzik, 2024-04-19:
Yeah Enrollment do not work with IOS 17.4.1 - iPhone 12 here and Apple Watch

bzumik1, 2024-04-19:
Did someone had a chance to test it on earlier version of iOS? Is it really issue with the latest SW.

jtowe1, 2024-04-19:
I got a different esp board and it worked.

luzik, 2024-04-19:
I am using esp32-s3-devkitc-1

jtowe1, 2024-04-19:
The one I had the problem with was a 
```
LILYGO T-Display-S3 ESP32-S3 1.9 inch ST7789 LCD Display TTGO Development Board
```

The one that works for me is
```
ESP-WROOM-32 ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with Antenna RF AMP Filter AP STA Compatible with Arduino IDE (3PCS)
```

bzumik1, 2024-05-05:
I am having issue on **esp32 c3 super mini**.

WilliamBradshaw, 2024-05-11:
Also having the same issue on a esp32 c3 super mini

trollix, 2024-05-15:
Same Issue here with iphone 15

[image]

tunip, 2024-05-25:
Same issue on M5Stack C3U (esp32c3).

labrunning, 2024-05-28:
I'm also having problems enrolling my iPhone. 

Beacon: iPhone 15 Plus, iOS Version 17.4.1
Base Station: EST-ESP32-S3-DevKit with ESP32-S3-WROOM-1-N8R8 (x 3)
ESPresense Version: v3.3.2

The ESPresence beacons are working as I would expect, and with a considerable amount of testing I can get the name of my phone as an id (i.e. apple:iphone15-5) but I'm concerned this is not going to be permanent. I don't have a MAC to get an irk.

tunip, 2024-05-28:
@labrunning you can use temporary a "normal" esp32 (not a s3 or c3) for pairing your iPhone to get the IRK. Afterwards you can switch it off and/or remove the temporary ESP32 again.

labrunning, 2024-05-28:
@tunip you like these? [https://shop.everythingsmart.io/products/est-esp32-devkit-esp32-development-kit](https://shop.everythingsmart.io/products/est-esp32-devkit-esp32-development-kit) I bought the other ones which were slightly more expensive because it expressly said they were ideal for ESPresence, turns out this isn't actually the case. I also struggled to find any kind of enclosure for them that I could buy, so I may just switch to the regular ESP32s and see if I can find another project for the S3s.

tunip, 2024-05-28:
@labrunning yes, the ESP32-WROOM-32 based ones would work fine with ESPresense and for pairing with your iPhone. Hopefully the "pairing bug" with the S3/C3 devices are fixed in the future. After pairing, you can continue to use your ESP32-S3 device, as this is only a one-off task and the S3/C3 devices otherwise work great.

labrunning, 2024-05-28:
@tunip thanks I will give that a try.

uohlukas, 2024-05-30:
I've tried with the "standard" ESP32 and managed to successfully pair my iPhone and now have a working irk number which has made everything much easier. Thanks for the help.

labrunning, 2024-05-30:
I flashed espresence on to a "standard" ESP32 today and managed to successfully enrol my IPhone. I now have an irk number which is making things much easier. Thanks for the help.

Tuinslak, 2024-06-05:
> ESP32-WROOM-32

> this is only a one-off task and the S3/C3 devices otherwise work great.

Hey -- so that I understand correctly (as a new user; as I haven't been able to get any further): the pairing only has to happen once (on one device) across the entire network of ESPresence? It's not something that needs to be done on every individual board? Is it a manual process of copying the IRK across devices, or is automatic?

tunip, 2024-06-05:
> Hey -- so that I understand correctly (as a new user; as I haven't been able to get any further): the pairing only has to happen once (on one device) across the entire network of ESPresence? It's [...]

You are right. Pairing only has to happen once on one device across the entire network of ESPresense. It is only to get the IRK from your iPhone/iPad which is "stored" on your MQTT Broker. Each ESPresense node can then "read" this information via MQTT.

lachi3, 2024-06-10:
This may have already been solved above. But i have an old iPad that isn't on iOS 17, and i'm still getting the same result.

broyuken, 2024-07-29:
I'm unable to get the irk for my phones or watches on both ios17 and 18 beta.

raptile, 2024-08-17:
same here on with multiple iphones.

schmiegelt, 2024-08-19:
Same here, doesn't work with iOS 17.
Unfortunatly I do not have a Mac available, so I cannot directly use the alternative method of retrieving the IRK. Is there any service where you can spin up a Mac VM for an hour or two? The one's I found are rather expensive and not suitable for short-term usage.

slashinpdx, 2024-09-21:
Having the same problem with the iPhone 16 Pro.  Unable to use the mac lookup method as well.  :(

akcoder, 2024-11-27:
Same issues as everyone else. I was able to get both my watch and phone enrolled, but I had to downgrade one of my nodes to **v3.2.4**.

Device: ESP32-S3

yuriisk, 2024-12-09:
Same issue.

Was able to get irk using friend's mac. To do that with different iCloud account I needed to pair iphone from mac bluetooth settings and then extract irk from Keychain Access.

closed 2025-04-23 by DTTerastar

drax16, 2025-09-16:
I know it's late, but I'm still having the same issue @meetbryce had. It appears to be connected on my iPhone itself, but it refuses to enroll on the web-ui side of things. I have a mac, but I want to add other devices that are not connected to my Apple ID. Using an ESP32 C3

quietCoder500, 2025-10-01:
I am also having the exact same issue, but I have no access to a mac to get the irk.

PR #1655 (2025-04-23): iOS 17/18 Enroll Fixed (fixes #1379, #1181)
fixes #1379
fixes #1181

## Summary by CodeRabbit

- **New Features**
  - Encrypted reads are now required for heart rate and device information characteristics, enhancing data security.

- **Bug Fixes**
  - Heart rate notifications are now throttled to a maximum of 4 times per second, ensuring consistent delivery.

- **Style**
  - Improved debug logging for notifications and status updates, providing clearer feedback during BLE operations.