ESPresense/ESPresense #1182: GL-iNET GL-S10 v2.1 Severe Packet Loss

Coder84619, 2024-01-06:
### Describe the bug

With ESPresence 3.3.1, the GL-iNET GL-S10 v2.1 Ethernet port has severe packet loss (roughly 50%). The solution, as found by the ESPHome team, is to change clk_mode to GPIO0_IN. 

Reference: https://github.com/esphome/bluetooth-proxies/commit/655ebf74775e6b7bb3abe7c22ce31c3dfcfc0ffe

### Version

3.3.1

### Device (be specific)

GL-iNET GL-S10 v2.1

Matsuo3rd, 2024-01-18:
Weird. Fix I provided to address this specific issue was already available in for 3.3.1 (version you indicated you are using).
See https://github.com/ESPresense/ESPresense/blob/v3.3.1/lib/network/Network.h (line 151)

Coder84619, 2024-01-18:
Ping tests via ethernet cable range from 12% to 50% packet loss for v3.3.1.

[image]

closed 2024-08-01 by stale[bot]