ESPresense/ESPresense #1185: Sensor for iPhone switches radomly to not_home

fhold, 2024-01-07:
### Describe the bug

I can't get ESPresence to work realiably with my iPhone 15. It randomly switches to not_home for 15 minutes or even hours.
I have the impression when I unlock the phone the sensor switches back to the correct phone.
How can i troubleshoot this?

I have 3 ESP32 for just a small apartement so coverage should be good. 
The iPhone is paired via IRK

current firmare.

Any help is appreciated

N3rdix, 2024-01-11:
Have you checked what the single ESPs report via MQTT (device topics under espresense/devices)? 

Is the phone not reporting any values at all?

fhold, 2024-01-16:
thanks for helping out. I'm a step further. I updated to the latest version (beta) and it seems to get better.
Some things I suspect is, that if the ESPs are running too long (2 days or more) the behaviour worsens. So restarting them seems to improve.
Second my apple watch is still problematic, here the problem is that after the screen goes black it goes to not_home - clicking on the display brings her back immediatly. I don't know if that is a norrmal behaviour with an apple watch, but like this I cannot use it with ESPresence.

N3rdix, 2024-01-16:
> Second my apple watch is still problematic, here the problem is that after the screen goes black it goes to not_home - clicking on the display brings her back immediately. I don't know if that is a normal behaviour with an apple watch, but like this I cannot use it with ESPresence.

This is working pretty well with my Apple Watch 5 and Ultra 2, both turning their screen of. They keep reporting "home" and also the distance to the nearest ESPresense instance is updated correctly. So could this be a watch setting (battery saver)?
Have you checked if the ESPs report your devices regularly in the UI [http://<ESP_IP>/ui/#/fingerprints](http://ESP_IP/ui/#/fingerprints)

fhold, 2024-01-16:
yes i checked that and when they are showing as not_home they are not reported in the fingerprints anymore until i switch the display of the phone or watch back on. 

What is strange is, that i seem to be more of an edgecase. I really don't know what I could do to improve the situation :-(

vibraHome, 2024-01-29:
Same problem here.

lorenjz, 2024-03-03:
I have the same issue too.  I can however see my iphone and my watch on the fingerprint page.

CrazYoshi, 2024-03-29:
I have the same issue. My iphone is not shown anymore on the fingerprint and it is reported as not_home. It will popup immediately when the screen is turned on again.
I'm currently running ESPresence 3.3.2
Any advise?

vibraHome, 2024-03-29:
Seems like the problems started with ios 17.
Can someone confirm this?

fhold, 2024-03-29:
not sure but that seems plausible. I switched to another BLE Tracker and still have the problem. I suppose it is a security feature of apple that causes this trouble, but thats just a wild guess.

fmonday, 2024-03-31:
I have the same issue with two iPhone 15s: one always advertises bluetooth, the other stops when the screen locks. It looks like [the devs know about it and this was posted](https://github.com/ESPresense/ESPresense.github.io/pull/114/commits/0ee2084515c3b5b6456fb43dcd1b99a2c2a12632)

Max24341, 2024-06-09:
my device shows up in the fingerprints but in HA its no_home. anyone knows a fix?
i made a alias for my device to get it to use like this.

  - platform: mqtt_room
    device_id: "iphone_15pm"
    name: "xxxx"
    state_topic: "espresense/devices/iphone_15pm"
    timeout: 10
    away_timeout: 30

closed 2024-12-21 by stale[bot]