ESPresense/ESPresense #1192: Settings page disappears once setting an included ID in the filtering section

ph15h, 2024-01-10:
### Describe the bug

Adding anything in this field 
`Include only sending these ids to mqtt (eg. apple:iphone10-6 apple:iphone13-2):`
removes the entire settings page located on `deviceIP/ui/#/settings` and makes the page blank. The HTML shows that it's
[image]

Other ESPresense devices settings pages are still fine. Sending an empty MQTT payload to `espresense/rooms/bedroom/include` to remove the IRK does not remove the setting. Restarting the device does not solve the issue and it shows up again. Reinstalling ESPresense entirely is the only 

Screenshot showing what's shown throught MQTT explorer
[image]

Affected ESPresense Device 

Happens on Chrome, Firefox and Edge.
[image]

Unaffected Device with that field empty
[image]

### Version

ESPresense Version: v3.3.1

Psythian, 2024-01-10:
In my testing when I put .5 instead of 0.5 and I changed to any other page from settings and went back it would immediately show the grey square. As long as I put the 0 in the front of the point it would work. So far after erasing and reloading several times that is my fix. Now to wipe and reload 4 more...
[image]

DTTerastar, 2024-01-20:
It doesn't happen for me.  I've always had stuff in include:

[image]

DTTerastar, 2024-01-20:
I can confirm that .5 causes issues:

SyntaxError: Unexpected token '.', ..."kip_dist":.5,"skip_m"... is not valid JSON

closed 2024-01-21 by DTTerastar

DTTerastar, 2024-01-21:
I released this in 3.3.2, please test!

commit dba3e9f0 (2024-01-21): Bump AsyncWiFiSettings to 1.0.10 (fixes #1192) (#1206)