ESPresense/ESPresense #1194: Frequent Reboots malloc failed to allocate

TommyJW, 2024-01-11:
### Describe the bug

I have 13 nodes. I observed ESPresense Companion was losing nodes in what appears to be a random fashion. Router uptime was also very low (in the order of seconds per node). Each node is assigned a static IP.
I pulled two nodes to collect logs and found they were reporting an error, then they would restart. I am operating under the assumption that is the cause of my observed symptom.

heap_caps_malloc was called but failed to allocate 205 bytes with 0x1800 capabilities. 

### Version

3.3.1

### Active scan enabled

unknown

### Include filter

None

### Exclude filter

None

### Query filter

Unknown

### Firmware flavor

Standard

### Device (be specific)

Teyleten Robot ESP 32 ESP-WROOM-32 30pin

### Logs

```text
FIRST DEVICE:
...........................................................................................................192.168.50.122
Firmware:     esp32
Version:      v3.3.1
WiFi BSSID:   04:D9:F5:FA:D7:80 (channel=11 rssi=-49)
IP address:   192.168.50.122
DNS address:  192.168.50.1
Hostname:     espresense-murphy_2
Room:         murphy_2
Mqtt server:  192.168.50.100:1883
Max Distance: 0.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 147148

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 New    | 11 Lines of New												            -61dBm 
...
1 Button | Button One: no Button Two: no
1 New    | 76 lines of New					                                        -48dBm 2098cb6870
[... 214 lines cut ...]
Max Distance: 0.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 147292

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 New    | 18 Lines of New            					                            -74dBm 3e1f0452356228
...
1 Button | Button One: no Button Two: no
1 New    | 67 Lines of New            					                            -91dBm 
...
1 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-88-7503 to node:gbath
```

geoffreycross, 2024-01-26:
Some issue for me.

`09:57:30.186 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.283 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.283 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.283 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.283 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.382 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.382 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.382 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.413 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.414 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.414 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.446 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.446 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.446 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.446 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.446 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.478 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.478 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.511 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.576 -> heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
09:57:30.640 -> heap_caps_malloc was called but failed to allocate 141 bytes with 0x1800 capabilities. 
09:57:30.673 -> 
09:57:30.673 -> abort() was called at PC 0x401c2862 on core 1
09:57:30.673 -> 
09:57:30.673 -> 
09:57:30.673 -> Backtrace: 0x40084041:0x3ffcee60 0x40096ad1:0x3ffcee80 0x4009a5b1:0x3ffceea0 0x401c2862:0x3ffcef20 0x401a821b:0x3ffcef40 0x401a8182:0x3ffcef60 0x400ef45d:0x3ffcef80 0x400f1558:0x3ffcefa0 0x400d50d6:0x3ffcefc0 0x401053c2:0x3ffcf000 0x401058de:0x3ffcf090 0x400e5755:0x3ffcf160
09:57:30.707 -> `

thomastodon, 2024-02-25:
I am also encountering this!

```
Firmware:     esp32
Version:      v3.3.2
```

```
heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 1096 bytes with 0x1800 capabilities.
```

It happens every 2-3 minutes on both of my ESP32 base stations

vcdx71, 2024-07-02:
Same here, anyone figure this out?

zanyraspi, 2024-09-29:
same issue with me. :(

bditter, 2024-10-02:
same for me.

heap_caps_malloc was called but failed to allocate 1696 bytes with 0x1800 capabilities. 
heap_caps_malloc was called but failed to allocate 170 bytes with 0x1800 capabilities. 

abort() was called at PC 0x401c2a1e on core 1

Backtrace: 0x40084041:0x3ffcede0 0x40096ad1:0x3ffcee00 0x4009a5b1:0x3ffcee20 0x401c2a1e:0x3ffceea0 0x401a83d7:0x3ffceec0 0x401a833e:0x3ffceee0 0x400ef961:0x3ffcef00 0x400f2fc4:0x3ffcef20 0x400d54f2:0x3ffcef40 0x40105c0e:0x3ffcef80 0x40105c8e:0x3ffcf010 0x400e5b7d:0x3ffcf0e0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

jsnyder, 2024-12-15:
I have this as well, is this essentially running out of RAM on the device?

closed 2025-06-27 by stale[bot]