ESPresense/ESPresense #1197: LEDs GPIO pin selection limited to GPIO39

gargamelonly, 2024-01-12:
### Describe the bug

I have an ESP32-S3 with up to GPIO48.
The configuration of ESPresense is limited only up to GPIO39.
It needs to be able to add the LED also up to GPIO48.

### Version

3.3.1

### Device (be specific)

ESP32-S3-WROOM-1

nathanedudley, 2024-04-17:
Same issue on 3.3.2 and 3.3.2b4, using the ESP32-S3-DEVKIT-C,

cdheiser, 2024-10-19:
This limit still exists, and it's difficult to submit a PR since there's little to no documentation about how to generate the UI code that is in github.

cooljimy84, 2024-12-20:
Just hit this one as well with my ESP32 S3 WROOM, has a RGB that i can't use due to being gpio 48

flo501, 2025-12-07:
Limit is still there. Can't use GPIO 48

Anoxym, 2025-12-18:
Okay, I found a workaround: you need to click on the field → inspect the element and remove the attribute max="39".

[image]