ESPresense/ESPresense #1198: Devices randomly show up in random rooms when given device is not even around

horvathakos723, 2024-01-13:
### Describe the bug

I have 3 ESP32's set up with ESPresense, and 4 iPhones configured through enrolling, discovery works fine, for more than a year now.
But sometimes, a device shows up in MQTT when it is not physically here. It happened a few times earlier, but lately it became quite frequent, more than once a week. Since they were configured through enrolling, and got the unique irk's, I guess neighbours' stuff is out of the question.
Restarting the ESP32 doesn't solve it, iphone still shows active in the room. Pressing 'Update' solves the issue, and also cutting the power solves the issue, but that's a faff.
I don't really have a custom setup other than the 'Include filter' which shouldn't cause this, and other than that, everything is default.
Any help is appreciated.

### Version

v3.3.1, but happened before as well

### Active scan enabled

is that still a thing?

### Include filter

enabled

### Exclude filter

-

### Query filter

-

### Firmware flavor

-

### Device (be specific)

esp-wroom-32, iPhone 13 pro, iPhone 13 mini

Ruben40870, 2024-01-23:
I am having the same issue. At the moment just my iPhone (iPhone 14 Pro Max) does this. When I leave the house the ESPresence will report my device present in the last known area. It has even switched rooms in the past while I have been away. It does every single time for me.

This completely defeats the purpose of using these devices to determine whether I am home or not.

I use the ESP32 D1 Mini, also three of them. Using ESPresence software version 3.3.2.

I hope the issue is resolved soon.

horvathakos723, 2024-01-23:
To be precise, in my case, when phones enter and leave the area, ESPresense works as expected, but sometimes devices randomly pop up out of nowhere.

Ruben40870, 2024-01-23:
Maybe I should then open a separate issue.

closed 2024-08-04 by stale[bot]