ESPresense/ESPresense #1199: Name device when enrolling devices

Steph1116, 2024-01-13:
### Describe the bug

Hi, 

I'm not sure if this is a bug or a one off. its quite annoying though as the device I was trying to add, belongs to my roomate and he and his phone are almost always gone to work or sleeping. I can't easily intercept him and fix it. 

Anyhow the problem is just the naming. I entered his "name iphone" when you click enroll, it then sends the bluetooth ESPresense to the phone.  I clicked it, paired it, and it said paired. I then looked on the list and just got the irk:34532554..... number and his alias and name was blank. I then "forget this device" on the iphone, went back to enroll, typed the "{name iphone" again and did the pairing. Again it paired but I'm not seeing the alias or name. So that's the issue.

I added 3 iphones and one watch and his phone is the only one with this issue as they all have the name and alias. I added his irk:32456343.... number to HA and it tracks it just fine. So obviously its just me being OCD that everything has the name I entered and his doesn't. 

Now, what can I do to completely remove his phone from ESPresense detection, and HA, and then try again? I mean, I know I can "forget device" on his phone, delete his irk: number from sensors.yaml in HA, but if I turn off his bluetooth and reset the phone and turn it back on, ESPresense still shows his ID on the enrolled devices list. Is there a way to, for example, turn off his bluetooth on his device to then delete the enrolled device from ESPresense memory?

Can someone make sense of this for me. I'm not in a rush as the phone, like I said, is connected, its just my OCD to have it right and organized. But it could also be a bug and potentially be an actual problem for someone else. 

Thanks

### Version

v3.3.1

### Active scan enabled

yes

### Device (be specific)

ESP32-D0WD-V3 times 2, ESP32-S3 times 2

PedroLageTavares, 2024-01-16:
Read in another thread that you can delete the enrolled device by deleting it from MQTT topic Settings and Device.

ripvega, 2024-03-15:
You can also set/change the name by publishing to the MQTT topic.

closed 2024-09-28 by stale[bot]

wanghuangjie, 2024-10-06:
https://github.com/ESPresense/ESPresense/discussions/1247#discussioncomment-10857841