ESPresense/ESPresense #1215: Devices removed and require re-enrollment

ph15h, 2024-01-27:
### Describe the bug

Every so often (sometimes within 2 days and sometimes in 2 weeks) all non-beacon devices get removed and require re-pairing and re-adding. It removes Pixel 7,  iPhone 15 Pro, and Apple Watch S8 and. They're all removed so it's not just an Apple only device issue. 

No other settings get reset.  Would normally not be an issue but enrolling and putting the device name into Home Assistant often requires remembering exact naming or renaming devices to match names again. I have 3 ESP32 devices in my home running ESPresense and the devices get removed from all 3

[image]

### Version

3.3.2

### Active scan enabled

No

hdurdle, 2024-02-03:
Also seeing this on 3.3.1 and 3.3.2.  Apple iPhone, Apple Watch devices are being removed - sometimes within 24 hours - while other iBeacon devices are retained.

oriolism, 2024-02-15:
Same issue here with iPhones.

ph15h, 2024-02-15:
I ended up just forgoing the device re-enrollment and utilized the IRKs as both the device_IDs so they wouldn't need to be re-enrolled any more should they disappear again

[image]
You do have to add the BLE IRKs to the scanning portion, though to get it to work.

[image]

rmiles7721, 2024-02-29:
Same issue with my Pixel 8 Pro and Pixel 7a

MHeuvel, 2024-03-04:
I have a similar issue with my iphone 14 using irk. I do notice that happens when my MQTT server is restarted and I need to re-enroll my device.

oriolism, 2024-04-25:
> Same issue with iPhone. 3 times enrolled, 3 times disappeared.

Try the recommendation from @ph15h, that fixed the issue for me.

dktrekkie, 2024-04-30:
I have experienced the exact same bug for some time now. My MQTT server is mosquito on a docker. It has been restated quite a few times. I will investigate if restarting that, is somehow related to this issue.

dktrekkie, 2024-05-04:
Confirmed. Issue seems linked to MQTT server restarts or downtime IMO. I re-added my iPhone, then restarted my Docker MQTT server (mosquitto). iPhone initially stayed in the list, but vanished after trying a esp-presence restart while server was down.

henriklied, 2024-06-26:
Experiencing the same issue as @dktrekkie and others

slashinpdx, 2024-09-23:
> I ended up just forgoing the device re-enrollment and utilized the IRKs as both the device_IDs so they wouldn't need to be re-enrolled any more should they disappear again [...]

How did you get the irk from you 15ProMax?

closed 2025-04-07 by stale[bot]