ESPresense/ESPresense #1225: Settings for immediate reaction?

Mikescotland, 2024-02-07:
### Is your feature request related to a problem? Please describe.

What would be the best settings for instant reaction for discovery of a beacon?
I'm looking to trigger presence in the shortest time possible, currently I can see it takes around 2 minutes for beacon to be discovered despite having HA app set to frequent beacon sent?

pickonedev, 2024-02-07:
This is how ble it is working... There is no instant reaction for ble... If you need something very fast, try a mmwave sensor (for human detection, not phone)

Mikescotland, 2024-02-07:
Need something that picks up exactly this device, radar isn't suitable for this purpose.
Bit puzzled because my phone sends beacon every second now, so there must be a way espresense picks it up immediately. I'm not bothered with which node picks it up. It's just for automation "when X person approaches the house"

gregschroeder, 2026-10-06:
I've submitted PR [#2520](https://github.com/ESPresense/ESPresense/pull/2520) (untested) as an attempt to mitigate this somewhat. A major culprit in the delay seems to be the forced smoothing over 15 seconds. This change would allow this to be a configurable parameter.

gregschroeder, 2026-10-11:
Tested with an alpha build with this change - works great!  Set smoothing window low (eg, 1000), and get near immediate updates.  For my use case, I don't care about any jitter or lack of smoothness, so this solves it for me, and hopefully others.

DTTerastar, 2026-10-31:
Resolved by #2520, available in the v5 betas ([v5.0.0b5](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b5)). The 15 second RSSI smoothing window is now the `rssi_window_ms` Filtering setting (100 to 60000 ms, default 15000). Lower values react much faster but are noisier, and it was tested in this thread with a value of 1000. Phones that throttle their own advertising can still add delay, which firmware can't remove.

closed 2026-10-31 by DTTerastar