ESPresense/ESPresense #1227: esp32-s3 won't start correctly without serial console

radkokeves, 2024-02-10:
### Describe the bug

replace build flags
  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1
with
  -D ARDUINO_USB_MODE=0
  -D ARDUINO_USB_CDC_ON_BOOT=0

https://github.com/espressif/arduino-esp32/issues/7554

tested with:
Waveshare ESP32-S3 Mini

### Device (be specific)

esp32-s3

borodaaaaa, 2024-02-12:
I wonder if that is related to not being able to get it working using a wall charger, only works when connected to a PC USB, not a power brick

radkokeves, 2024-02-12:
@borodaaaaa  exactly, that is the way how to reproduce it

borodaaaaa, 2024-02-12:
Wow, cool that you got to the root cause of it. Would be great if that can be fixed.

borodaaaaa, 2024-02-12:
Thank you for that mate, unfortunately i am super noob with esp…  i flashed it via browser as per the doco. How do i access that file ? I have web gui of espresense i know the ip esp…  i guess if i can mount its fs somehow ? Or how do i get access to that file ?  I have eap wroom 32 if that changes anything

radkokeves, 2024-02-12:
@borodaaaaa, when flashing via web, there is dropdown (flavor) allowing you to choose build (env).
Currently there is no build OOTB with configuration flags needed for it to work.
You need someone to test my hypothesis (and/or) create new flavor (build) for the website with build flags as i wrote in previous comment

borodaaaaa, 2024-02-12:
Ok. So ot is something that cannot be fixed easily. Thank you for making it a bit clearer. Hope the owner looks at that and find a workaround.

radkokeves, 2024-02-12:
@borodaaaaa it can be done easily by maintainer of repository (if it is correct solution),
 or someone can build it for you and provide you with image, but not sure, if you want to flash image from stranger

Wildbill-Z, 2024-02-14:
See also [HERE](https://github.com/ESPresense/ESPresense/issues/1131)

radkokeves, 2024-02-14:
@Wildbill-Z  i don't want to stress out difference in difficulty for end user: 
solution you provided is:  power on from any 5V source when either holding down "BOOT" while plugging it in or by pulling down GPIO0 to ground with a 10k resistor - every boot, in case of power outage it would not recover by itself...

solution I provided is: recompile it with different flags (probably different flavor/env) to be more transparent for end user

Wildbill-Z, 2024-02-14:
@radkokeves I didn‘t provide any solution. I just linked to the other issue about the same problem so in each of them you‘ll see the other as they have the same cause. I also think, the better (best) solution is a compiled image with different flags as you mentioned.

radkokeves, 2024-02-14:
@Wildbill-Z i understand, thank you for clarification and sorry for tone of my previous response

perever, 2024-04-23:
this is definitely a hardware problem. I have a couple ESPs and one has the same problem. I power it via 3.3V and the GND pins and everything works fine.

Wildbill-Z, 2024-04-23:
@perever Sorry, but it is definitly no hardware problem. All of my devices just work perfect by downgrading to version 3.2.2 as they did before I updated them to the latest version where all of them (5) had this problem. And as @radkokeves wrote above, by changing settings in platform.ini it is also solved. And you won‘t „repair“ a hardware problem by changing some settings in a file.

perever, 2024-04-23:
this bug doesnt  depend on any .ini or other files. it is independent of ESPresence. You can completely erase ESP and flash any example from the Arduino IDE. Everything will work fine. EXCEPT Wi-Fi.

Wildbill-Z, 2024-04-23:
Didn‘t you say, you think it is a hardware problem? How would it be that it is a problem with any example from Arduino IDE with every device? And how would it be, that older versions of Espresence don‘t have the problem? Exactly! It is a change in a library within Arduino IDE which Espresence also uses. A software problem! Older versions of Espresence did work well, newer versions which use the newer version have the problem. So I went back to Espresence 3.2.2 which uses the older version and all is working fine again. Flash a newer version and you will only (for >90% of all tries) have the effect, that plugging the device to just power it won‘t connect to Wifi but connecting it to your computer via USB it will work all the time. This is the effect of the change in the library and this is what this thread and problem is all about. Believe it or not.

perever, 2024-04-23:
as I mention above, I had this problem. The solution, that helps me is to power ESP via 3.3V and GND pins.

borodaaaaa, 2024-04-23:
Not sure about cannot connect to wifi, my esps go offline in home assistant but i still can ping them but with packet loss and web gui  does not load. A reboot fixes it for sometime… it is not coverage issue  all have 2.4 only nets and rssi is another 60-70-80

perever, 2024-04-23:
what webserver library do you use ?

perever, 2024-04-24:
I have had similar cases. in my case it was after a temporary loss of Wi-Fi signal due to router reboot. ping is ok, but the server GUI is not responding. The only thing that helped was a reboot.

borodaaaaa, 2024-04-25:
Yep reboot of esp32 helps. I dont think it is due to the router as the only device that i touch (reboot) is the esp. No issues for the other wireless clients

perever, 2024-04-25:
sometimes i manualy reboot router.  so i added a wifi watchdog to  the esp, that reboot  it on wifi loss  . the problem is gone.

borodaaaaa, 2024-04-25:
What do you mean gone ? When  it reboots it does not register BLE nor ut pushes mqtt updates. It is useless and it breaks automations

helto4real, 2024-07-12:
I have the same problem, latest version of ESPrecense

closed 2025-01-24 by stale[bot]