ESPresense/ESPresense #1237: Status LED

sEdsKi, 2024-02-24:
### Describe the bug

Documentation references a checkbox to disable the status LED. This option is not present ;-)

[image]

### Version

ESPresense Version: v3.3.1

### Device (be specific)

ESP32 WROOM 32

CoMPaTech, 2024-02-28:
I think I came across this before but if it's a LED like the blue one on mine it can't be disabled programmatically if I recall correctly.

ginkel, 2024-03-06:
I think you can disable the blue LED by setting the LED 1 pin to `-1`.

Oli-BY, 2024-03-17:
There isnt any possiblity to switch the red ESP32 LED off. That would be very helpful

ginkel, 2024-03-17:
The red LED is always on, afaik. You can unsolder it if it annoys you.

closed 2024-09-29 by stale[bot]