ESPresense/ESPresense #1254: ESP32 WRoom with sen0395

wifispray, 2024-03-14:
### Describe the bug

Can someone help with both the config in Esprence GUI and the pinouts used. I cant seem to work it out.

wifispray, 2024-03-14:
This doesnt mean anything to me.... A noob! Radar Motion
The Radar motion sensor can be configured in the GPIO section. There are several options on how to configure the pin to match the specific Radar device in use.

PIR motion pin type:
Pullup
Pullup Inverted
Pulldown
Pulldown Inverted
Floating
Floating Inverted
PIR motion pin (-1 for disable):
PIR motion timeout (in seconds):

erdie721, 2024-06-21:
Supply 5v and ground, then whatever you connect to IO2 on the sensor, set that number in ESPresense under Radar. Motion Pin Type should be "pull-up".

closed 2025-01-01 by stale[bot]