ESPresense/ESPresense #1256: MQTT connection times out every 60 seconds due to TCP

leccelecce, 2024-03-19:
### Describe the bug

According to my MQTT broker (Mosquitto) and ESPResence's own logs, the MQTT connection is being dropped every 60 seconds with status 0 which seems to indicate "TCP failure".

The ESP32 is not rebooting, as the uptime figure continue increasing. Free memory hovers around 82k, never below 60k.

I have 6 of the same device, all running the same firmware/settings, and they all behave identically.

Looking in my firewall logs, it seems that the firewall is seeing a lot of TCP packets arrive out of order, suggesting possibly some issue in the Wifi or TCP stack?

### Version

3.3.2

### Active scan enabled

no

### Include filter

iBeacon:

### Exclude filter

n/a

### Query filter

n/a

### Firmware flavor

esp32

### Device (be specific)

ESP32-WROOM-32E

### Logs

```text
Logs from ESP32 USB terminal:

Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...

... config stuff ...

Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...

... config stuff ...

Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...

Logs from Mosquitto 2.0.18 - repeated every ~60 seconds indefinitely:

2024-03-19T17:45:28: New client connected from 10.0.0.2:56366 as espresense-office (p2, c1, k15).
2024-03-19T17:45:28: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:45:28: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:45:28: espresense-office 1 espresense/settings/+/config
2024-03-19T17:46:31: Client espresense-office has exceeded timeout, disconnecting.
2024-03-19T17:46:34: New client connected from 10.0.0.2:56367 as espresense-office (p2, c1, k15).
2024-03-19T17:46:34: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:46:34: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:46:34: espresense-office 1 espresense/settings/+/config
2024-03-19T17:47:31: Client espresense-office has exceeded timeout, disconnecting.
2024-03-19T17:47:34: New client connected from 10.0.0.2:56368 as espresense-office (p2, c1, k15).
2024-03-19T17:47:34: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:47:34: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:47:34: espresense-office 1 espresense/settings/+/config
2024-03-19T17:48:25: Client espresense-office has exceeded timeout, disconnecting.
2024-03-19T17:48:28: New client connected from 10.0.0.2:56369 as espresense-office (p2, c1, k15).
2024-03-19T17:48:28: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:48:28: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:48:28: espresense-office 1 espresense/settings/+/config
2024-03-19T17:49:31: Client espresense-office has exceeded timeout, disconnecting.
2024-03-19T17:49:34: New client connected from 10.0.0.2:56370 as espresense-office (p2, c1, k15).
2024-03-19T17:49:34: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:49:34: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:49:34: espresense-office 1 espresense/settings/+/config
2024-03-19T17:50:37: Client espresense-office has exceeded timeout, disconnecting.
2024-03-19T17:50:40: New client connected from 10.0.0.2:56371 as espresense-office (p2, c1, k15).
2024-03-19T17:50:40: espresense-office 1 espresense/rooms/*/+/set
2024-03-19T17:50:40: espresense-office 1 espresense/rooms/office/+/set
2024-03-19T17:50:40: espresense-office 1 espresense/settings/+/config
2024-03-19T17:51:37: Client espresense-office has exceeded timeout, disconnecting.
```

alexmoras, 2024-03-23:
Exactly the same issue here with the ESP32-WROOM-32. Exactly 60 seconds of connection before TCP is dropped and it reconnects to Mosquito.

Issue clearly lies with the ESP32 device since I have quite a large MQTT network and no other devices experience this issue. I have another ESP32 device running ESPHome and using MQTT - this is functioning fine.

leccelecce, 2024-03-24:
I have turned on debug logging on Mosquitto, which is showing some interesting data. I've filtered on 'PING' and 'disconnecting'.

```
2024-03-24T15:11:38: Received PINGREQ from espresense-office
2024-03-24T15:11:38: Sending PINGRESP to espresense-office
2024-03-24T15:11:49: Received PINGREQ from espresense-office
2024-03-24T15:11:49: Sending PINGRESP to espresense-office
2024-03-24T15:11:59: Received PINGREQ from espresense-office
2024-03-24T15:11:59: Sending PINGRESP to espresense-office
2024-03-24T15:12:24: Client espresense-office has exceeded timeout, disconnecting.
2024-03-24T15:12:38: Received PINGREQ from espresense-office
2024-03-24T15:12:38: Sending PINGRESP to espresense-office
2024-03-24T15:12:49: Received PINGREQ from espresense-office
2024-03-24T15:12:49: Sending PINGRESP to espresense-office
2024-03-24T15:13:24: Client espresense-office has exceeded timeout, disconnecting.
2024-03-24T15:13:38: Received PINGREQ from espresense-office
2024-03-24T15:13:38: Sending PINGRESP to espresense-office
2024-03-24T15:13:48: Received PINGREQ from espresense-office
2024-03-24T15:13:48: Sending PINGRESP to espresense-office
2024-03-24T15:13:59: Received PINGREQ from espresense-office
2024-03-24T15:13:59: Sending PINGRESP to espresense-office
2024-03-24T15:14:24: Client espresense-office has exceeded timeout, disconnecting.
2024-03-24T15:14:38: Received PINGREQ from espresense-office
2024-03-24T15:14:38: Sending PINGRESP to espresense-office
2024-03-24T15:14:49: Received PINGREQ from espresense-office
2024-03-24T15:14:49: Sending PINGRESP to espresense-office
2024-03-24T15:15:00: Received PINGREQ from espresense-office
2024-03-24T15:15:00: Sending PINGRESP to espresense-office
2024-03-24T15:15:24: Client espresense-office has exceeded timeout, disconnecting.
```

It looks like ESPresence is *usually* sending a ping around every 11 seconds. Given the default timeout is 15 seconds, this should normally be sufficient.

However, every 3rd or 4th table it doesn't seem to send a ping, with the result that the broker then terminates the connection. 

From what I can see, ESPresence doesn't explicitly send pings but leaves it to the underlying library (AsyncMqttClient) here. You can see it sends a ping every `timeout * 0.7` (giving 10.5) which matches up to the ~11 seconds.

https://github.com/marvinroger/async-mqtt-client/blob/89bf46485d5b60ce1e8e5e4d265a9c1570de3dc5/src/AsyncMqttClient.cpp#L331

What isn't clear is what is calling the the onPoll() method that checks whether to send a ping. [This comment](https://github.com/marvinroger/async-mqtt-client/issues/172#issuecomment-568173689) suggests it's the underlying AsyncTCP library and it should be called every 125ms - question now is to find out whether that's actually happening or not.

The other possibility is that the AsyncMqttClient isn't sending a ping because it thinks there are other messages being sent anyway - next step in debugging is to look for that.

leccelecce, 2024-03-24:
Looking at the AsyncMqttClient library, worryingly it's had no commits since 2022.

The current published release (used by ESPresense) was released in 2021. Since then it looks like there was an attempt to fix the ping code in https://github.com/marvinroger/async-mqtt-client/pull/284 but there is no released version with this code.

Looking at that library, and ASyncTCP, it seems like there have been various issues over time and various half-abandoned forks. Unfortunately I'm not sure where that leaves this because it would be non-trivial to swap out the library from ESPresence. My C++ / TCP / hardware build/debugging is also not good enough to figure out where in the stack this is going wrong without investing a load more time.

leccelecce, 2024-04-16:
I don't suppose any maintainers with familiarity with these areas of code could comment?

leccelecce, 2024-04-29:
Looking at ESPHome, it seems they are using a separately maintained version of async-mqtt-client (here: https://github.com/HeMan/async-mqtt-client), given the marvinroger version used by ESPresence looks to be abandoned.

Given that it seems ESPresence is not keeping up with updates in underlying libraries, reluctantly I will have to switch to ESPHome. A shame as this app is all but perfect for me and the Companion looks seriously cool, but 30 second drop-outs on my hardware make it difficult to tune.

J0han3s, 2024-06-21:
any comments/updates on this?

leccelecce, 2024-06-21:
Yes - my issue was not with ESPresence. I tried OpenMqttGateway and Esphome and had similar issues, so investigated my network and just found my router was dropping the MQTT ping packets when routing across VLANs/subnets, possibly due to some odd behaviour in the esp wifi stack. I solved the problem by moving the wifi used to the same subnet as Mosquitto, but there are other areas to check as well (WiFi Ap related in most cases)

closed 2024-06-25 by leccelecce

J0han3s, 2024-06-26:
Cheers, helpful. Seems I might be having the same issue. What router WAP you using - how do I confirm if packets are being dropped ?

leccelecce, 2024-06-26:
I have Unifi U6 AP's, connected to a variety of Netgear and Mikrotik managed switches, all ultimately going back to a x86 box running pfSense CE 2.7. I have an "IOT" wifi SSID linked to VLAN30/10.10.30.x, with my Mosquitto instance running in VLAN20/10.10.20.x. PfSense is doing the inter-VLAN routing.

Turning on debug logging on the ESP32s and Mosquitto _seemed_ to proved that the ESP32 was sending a PING, Mosquitto was receiving the PING and responding, but the ESP32 was never receiving the response and so was deeming the connection to be dead. When it then reconnected, Mosquitto was surprised to see a new connection (as it thought the old one was still alive).

I was in the midst of trying to look at packet captures at various points to see if I could prove that Mosquitto was genuinely sending PING responses back to the ESP32s, when I came across a Reddit comment that suggested possibly ARP responses or routing tables were the problem i.e. the PING packet being sent, but a network device somewhere dropping it. I had spent a bit of time loosening various firewall rules to try and see if this was happening but couldn't see it occurring. As an experiement I tweaked my Mosquitto VM to also listen on the VLAN30/10.10.30.x network i.e. no cross-VLAN or subnet routing required, and lo and behold my problem disappeared. I don't know whether it was an ARP/layer 2 problem or an IP/layer 3 problem and haven't had time to dig into it further.