ESPresense/ESPresense #1265: Hidden 40 MHz wide SSID broadcast whilst connected to main AP anyway - request disable local AP mode entirely

nzkiwi68, 2024-04-01:
### Is your feature request related to a problem? Please describe.

If your main AP goes offline, then the ESP start broadcasting an open unprotected SSID.
If the ESP is properly connected to your AP, it remains broadcasting a hidden SSID anyway.

### Describe the solution you'd like

I would like to see an option to completely disable the ESP "AP mode" for security.

I realise that this would mean if I needed to get access to the ESP and for any reason and it will not longer connect to my wireless AP, I would need to reflash it and start again. I'm ok with that.

### Additional context

This improves the security.

Nobody has to enable the proposed new "switch off" AP mode, only if they want to, and, I want to.

peter-dolkens, 2024-07-29:
I believe this is two tickets:

* Bug report which I suspect is invalid. Do you have unifi network gear by any chance? Very few devices can operate in AP mode at the same time they're operating in client mode, let alone a cheap ESP device.
* Feature request which is a duplicate.

ryanrozich, 2024-08-30:
I am also seeing my espresence devices broadcasting ssids while connected to my APs. I do have unifi equipment. Is this a known issue with unifi?

I have 8 espresence devices and am concerned that they might be clogging up the 2.4G spectrum.

ryanrozich, 2024-08-30:
Here is one example showing the device connected to my unifi Living Room AP and also showing up as broadcasting close to it as well

[image]

[image]

DTTerastar, 2025-02-08:
Seconds to wait for WiFi before captive portal if set to -1 should never pop the AP.

DTTerastar, 2025-02-08:
And I think the reason you see that in the scan is the AP has to disconnect everyone to do the scan.

peter-dolkens, 2025-02-09:
Can confirm - I'm running dozens of ESP devices on a farm, hundreds of meters away from anyone. I can perform scans without causing them to disconnect, and there is indeed no rogue APs.

DTTerastar, 2026-11-07:
Sorry for the bot noise earlier: #2161 was closed without merging. This is covered in v5, though, so I'm closing it.

- Once a v5 node joins your WiFi it runs in station-only mode, so it no longer broadcasts a hidden SSID while connected.
- If WiFi drops at runtime, the node keeps trying to reconnect and never opens the setup AP.
- To make sure the setup AP never opens, set **Seconds to wait for WiFi before captive portal** to `-1` on the Network page. The node then waits for your WiFi forever, even at boot. The only exception is a node with no WiFi configured at all (first setup).
- If you'd rather keep the AP as a fallback, set **Configuration AP Password** (8-63 characters) and it comes up as WPA2 instead of open.

All of this is in [v5.0.0b8](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b8). v5 changes some settings, MQTT topics and Home Assistant entities, so try it on one node first. If you still see the node broadcasting while it's connected on v5, reopen this with a screenshot and your firmware version.

closed 2026-11-07 by DTTerastar