ESPresense/ESPresense #1270: Support for ESP32-C6

rgnyldz, 2024-04-08:
### Is your feature request related to a problem? Please describe.

Currently ESPresense cannot be installed on an ESP32-C6 board. 

### Describe the solution you'd like

Add support for ESP32-c6 based boards.

### Additional context

[ESP32-C6](https://www.espressif.com/en/products/socs/esp32-c6)

tmycgn, 2024-06-12:
Awesome Bord would be the M5Stack M5NanoC6 with WiFi6, etc. https://shop.m5stack.com/products/m5stack-nanoc6-dev-kit 🚀

stefan-schweiger, 2024-06-13:
Just cloned the project and looked if there is some easy fix for this, but it seems like the ESP32 is only supported by Arduino-ESP32 3.x framework which contains quite a few breaking changes, so as long as ESPresence does not upgrade to this I don't think we will be getting support for the C6

hajar97, 2024-06-13:
When is the plan to upgrade to this framework, do you know? Just to have an idea if we may get support for ESP32-c6 boards sometime soon or not?

stefan-schweiger, 2024-06-13:
From reading this thread I think the issue is down more to a disagreement between platformIO and espressif and there is no resolution in sight currently

https://github.com/platformio/platform-espressif32/issues/1225

hajar97, 2024-06-13:
> platformio/platform-espressif32#1225 <https://github.com/platformio/platform-espressif32/issues/1225>

Oh I see. That’s a great pity. Platform based on ESP32-C6 (M5NanoC6) is both more powerful and has smaller form factor than all the alternatives.

Jason2866, 2024-06-17:
> From reading this thread I think the issue is down more to a disagreement between platformIO and espressif and there is no resolution in sight currently
> 
> [platformio/platform-espressif32#1225](https://github.com/platformio/platform-espressif32/issues/1225)

Not really since the project uses my Platformio framework build. We build Tasmota since a few month for the C6 already ;-)

Cirehtims, 2024-08-22:
What can we do to help this to be treated sooner ?
I just bought ESP32-C6 boards to switch from wifi to zigbee and extends my presence detection network.
Thanks

robmwalsh, 2024-10-04:
> What can we do to help this to be treated sooner ? I just bought ESP32-C6 boards to switch from wifi to zigbee and extends my presence detection network.
> 
> Same request on ESPHome. [esphome/issues#5864](https://github.com/esphome/issues/issues/5864)
> 
> Thanks

@Cirehtims It just needs someone prepared to do the work ;)

If you're interested in helping out, here's the PR which added C6 support to Tasmota (I think) 
https://github.com/arendst/Tasmota/pull/19315

A quick search shows some other stuff which may be relevant
https://github.com/arendst/Tasmota/pulls?q=is%3Apr+is%3Aclosed+c6

@Jason2866 would you be prepared to assist with guidance for someone who wanted to give this a crack (hopefully @Cirehtims?)

Jason2866, 2024-10-05:
Full support for C6 with orig. Arduino espressif core 3.0.5 is here
https://github.com/pioarduino/platform-espressif32

pateism, 2024-12-02:
> Full support for C6 with orig. Arduino espressif core 3.0.5 is here https://github.com/pioarduino/platform-espressif32

Shame this is build on top of Microsoft VB. As a non windows user it’s a non-runner for some of us

Jason2866, 2024-12-02:
@pateism What makes you think any MS product is needed? Please do not post if you don't know how it works.

pateism, 2024-12-02:
> @pateism What makes you think any MS product is needed? Please do not post if you don't know how it works.

From the link you posted:

[Download and install official Microsoft Visual Studio Code](https://code.visualstudio.com/). pioarduino IDE is built on top of it.
Open the extension manager.
Search for the pioarduino ide extension.
Install pioarduino IDE extension

Jason2866, 2024-12-02:
That's only when you want to use VSC as IDE.

PinguDEV-original, 2026-01-15:
I'd also love to see C6 support 👍

cdheiser, 2026-02-05:
I've been working at this for a bit and believe I have a working fork of ESPresense as well as a number of libaries it uses or has forked to get the esp32-c6 working.  At least, my ESP32-C6-DevKitC-1 is working.

I think there's a clean path to get everything in place, but it will mean a breaking change for most people as SPIFFS is dead.  Maybe there's a decent way to migrate from SPIFFS to LittleFs for the exiting boards.  I haven't tried to figure out if/how that's possible yet.

At the moment, it looks like roughly the following changes would be necessarry:

TFT_eSPI: Needs to get caught up to present day and re-patched for M5Stack and handling for SPIFFS vs. LittleFS
ESPresense: Update from the very old 2023 tasmota espressif32 platform to the January 2026 release (this is a big change where SPIFFS is no more). This also requires a move to h2zero/NimBLE-Arduino@^2.2.1 or updating the forked version of the library.  c6 support can be added as part of this step.

If you want to test out the changes, you can find them here: https://github.com/cdheiser/ESPresense/tree/esp32c6

DTTerastar, 2026-10-31:
ESP32-C6 support landed in #2181 and ships as `esp32c6` builds in the v5 betas ([v5.0.0b5](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b5)). It isn't in the v4.0.x stable releases.

closed 2026-10-31 by DTTerastar