ESPresense/ESPresense #1274: Radar sensor: Hi-Link LD2450

gkchim, 2024-04-14:
### Is your feature request related to a problem? Please describe.

Integrate Hi-Link Ld2450 radar. Utilises a serial connection and offer more features than ld2410. Specifically it reports number of people (upto 3) and also reports presence in different zones.

### Describe the solution you'd like

Integrate the sensor, similar to how the ld2410 is, but would need to be a serial connection ( I think)/ not just binary

baudneo, 2024-04-16:
This would be really great to have access to, Right now I have a separate ESP32-C3 for my LD2450. If ESPresense could handle it, that would be great!

gkchim, 2024-05-26:
Good to know if this might be possible please?

baudneo, 2024-05-28:
> Good to know if this might be possible please?

I dont think so. The current radar setup uses 1 pin as a 'motion / no motion' pin. The LD2410 has a pin out for this, the LD2450 does not have a pin out to signal 'motion / no motion'. 

They would need to add support via UART and I do not see that happening anytime soon.

jploonstra, 2024-06-13:
> I dont think so. The current radar setup uses 1 pin as a 'motion / no motion' pin. The LD2410 has a pin out for this, the LD2450 does not have a pin out to signal 'motion / no motion'.
>
> They would [...]

It may not be feasible on the short term, but it would be a great addition to the ESPresense platform!
This is another example: https://www.dfrobot.com/product-2795.html

DTTerastar, 2026-11-04:
#2611 adds read-only UART support for the LD2450: target count, and X/Y/speed/distance for each of the 3 targets, plus presence/moving/still. It hasn't been run against a real module yet, so testing reports are welcome.