ESPresense/ESPresense #1284: The ESP32-C3 Super Mini is not functioning properly.

diyso, 2024-05-02:
### Describe the bug

The firmware for the Super Mini has been successfully uploaded.
WiFi on the phone has been connected, and the router connection is complete.
Confirmed MQTT connection after connecting to the Espressense.

However, when Enroll is activated, it is not possible to confirm Bluetooth connection on the Super Mini.
I conducted tests using the ESP-WROOM-32.
Successfully received the IRK value.

In the case of the Super Mini, the counter continued to be maintained during Bluetooth connection.
However, in the case of the ESP-WROOM-32, when connected, it completed and displayed the result.

------------------------
I wrote with the help of a translation tool because I don't know English.
Thank you.

### Version

3.3.2

Tuinslak, 2024-06-05:
Facing the same issue.

The board itself can be purchased here: https://www.aliexpress.com/item/1005005967641936.html

jploonstra, 2024-06-06:
My ESP32-C3 Super Mini got flashed, but then it stops functioning completely. USB (serial) is not recognized. I had to completely reset it and now it gets recognized but:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x7 (DOWNLOAD(USB/UART0/1))
Saved PC:0x400462e2
waiting for download
Please help...

lachi3, 2024-06-10:
Has there been any updates? I have also got a ESP32-C3 Super mini, and i'm having the same issue.

Tuinslak, 2024-06-11:
Can confirm it's properly pairing with an ESP32-WROOM-32S (https://www.aliexpress.com/item/1005006336964908.html). So the issue is somewhere with the C3 (mini) or the code on these boards.

Hubseba, 2024-06-12:
> My ESP32-C3 Super Mini got flashed, but then it stops functioning completely. USB (serial) is not recognized. I had to completely reset it and now it gets recognized but: [...]
Same here

jploonstra, 2024-07-22:
Is there any work being done on this project anymore? It seems the list of issues is growing and nothing is being handled. It is a shame, because this presence thing is working great. But should I be looking for a different solution?

lj4o7, 2024-07-26:
After click “Enroll”,my ESP32-C3 Super Mini can not pair with my iphone 12mini,whoes irk has been aquired by a ESP32-C3 before.
But in fingerprints,sometimes this known irk is detected and listed by ESP32-C3 Super Mini,distance not so accurate though.

gimmel, 2024-09-08:
I can flash my ESP32-C3 dev board, but the installer doesn’t ask for the WiFi data. So I use the internal web server to edit the data, and it works. But it’s getting so hot that my first ESP32-C3 stopped working. The second one is getting hot too, so I think the device is not compatible. Too bad, would have been just the right size.

imbaczek, 2024-11-10:
can confirm they work after reset and connecting to the AP. also confirm they get a tad hot but nothing extreme

oxygala, 2024-11-11:
I could also connect to the AP but my Wear OS watch cannot secure pair with ESP32-C3.

Aziona, 2025-01-01:
I connected one esp32 without problems but another one could not connect to wifi: neither to my Unifi Ap or to my internet providers router. When I connected my old Linksys E3000 and disabled its 5Ghz band it connected without problems to the E3000. So the issue might be connected to the esp32 not being able to select the 2.4Ghz SSID if a 5.0Ghz SSID is broadcasting with the same name (or something like that). Is that possible? If so, is it solvable by changing some settings on the AP side?

Hubseba, 2025-01-02:
This is possible. By BMS (battery-management-system) of my home energy storage has the same problem

Aziona, 2025-01-02:
Somebody suggested to change the channel on the 2,4Ghz or change the SSID on it, which I should have considered myself as well. My router also had the possiblity to add another SSID and seemed to work as well. Did not yet try the other options. 
Anyway, I am glad I can get it working now as it seems to work quite nicely in the few days that it has been working now.