ESPresense/ESPresense #1286: RSSI@1m iBeacon

MeierAlex, 2024-05-05:
### Describe the bug

There is a problem with the RSSI@1m which gets transmitted by the Tag as -60dBm but is processed as +60dBm by Espresence leading to wrong distance readings.

### Version

3.3.1

### Device (be specific)

ESP32 Devkit C

husnainali09, 2024-05-17:
Holyiot iBeacon is also being read as positive rssi@1m instead of a negative value. Any fixes for this ?

husnainali09, 2024-05-21:
Any fixes or bandaids ?

mgomersbach, 2024-11-19:
Please reopen. this is still a issue

closed 2025-06-01 by stale[bot]