ESPresense/ESPresense #1297: ESP32-S3-Zero

CedjaY, 2024-06-01:
### Describe the bug

Hello,

I am a new user of ESPresense.
I managed to get it working on ESP32 WROOM. I read that it could work on ESP32-S3, however I was provided with ESP32-S3-Zero mini. This doesn't seem to work. The installation goes smoothly, but it does not start.

Is there any chance that this will work in the future?

Here are the specifics:

`Equipped with Xtensa® 32-bit LX7 dual-core processor, up to 240MHz main frequency
Supports 2.4GHz Wi-Fi (802.11 b/g/n) and Bluetooth® 5 (LE)
Built in 512KB of SRAM and 384KB ROM, onboard 4MB Flash memory and 2MB PSRAM
Castellated module and onboard ceramic antenna, allows soldering direct to carrier boards
Supports flexible clock, module power supply independent setting, and other controls to realize low power consumption in different scenarios
Integrated with USB serial port full-speed controller, 34 × GPIO pins allows flexibly configuring pin functions
4 × SPI, 2 × I2C, 3 × UART, 2 × I2S, 2 × ADC, etc.`

Works perfectly with ESP Web Tools demo.

Do not hesitate to tell me if you need any other information that I have omitted.

THANKS

### Version

3.3.2

### Device (be specific)

esp32s3-20210327

### Logs

```text
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f4736
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x2878
entry 0x403c98bc
[    45][E][esp32-hal-misc.c:258] initArduino(): Failed to initialize NVS! Error: 24579
```

CedjaY, 2024-06-12:
> see #1227

This topic indicates that there is a problem when it is not connected to the computer. It's not my case...!

JackPinni, 2024-06-28:
I have the same problem, tried it on 3 different esp32-s3-zero and all gave the same error

artkeller, 2024-07-31:
I think you may also have the following problem, which may also be an issue with the ESP32-S3-ZERO boards that are currently (07/2024) being “sold off” at very low prices, but also with other mini boards with the same chip. The designation on the chip is FH4R2... which means: 4MB flash memory and 2 MB PSRAM. I tried for a long time to read the chip flashes with esptool (v4.7.0) and the command read_flash 0 ALL. After exactly 2MB I got a read error on all 10 boards. On suspicion I only read read_flash 0 0x200000 and lo and behold, it contained the complete program. After deleting the flash and writing it back with write_flash the program ran perfectly. I checked the PSRAM, it is completely present, only half of the flash is missing from this ESP chip of manufacturer Id=20.  

I was able to load my own programs when the partinioning had specified 1.3MB/0.7MB (=2MB).

Conclusion: mysterious memory error of all boards I bought with this chip (vendor id=20). Usable with 2MB flash partitioning. 

Cause; fraud? Quality defect? Rejects?

jmyoung, 2024-09-09:
Same problem here, with two Wemos Lolin S3-Mini.  Chip has FH4R2 on it, but I _can_ read back 4Mb of flash, so I don't think I have the problem @artkeller is having.  However I do get the same problem @CedjaY has.

If I reflash back with the original Micropython that the unit came with, I have access to Micropython again, it seems to be specificlaly the ESPresense build.  I've tried various versions to no effect.

vsviridov, 2024-11-09:
How does one check if it indeed has 4Mb of flash? I have bought a few WaveShare Modules off [Amazon](https://www.amazon.ca/dp/B0CR9YZHFH), and it fails to start... Tried with `v3.3.5` and `v3.3.5 verbose`.

arbigator, 2025-01-09:
@vsviridov Can you give more info on how you were able to flash? I've got a few of the WaveShare modules as well, but not having a lot of luck. I've got platformio in vs code, but it still does not work. Thanks!

vsviridov, 2025-01-09:
@arbigator I've actually since moved on to running Bermuda Trilateration + ESPHome BT Proxies, but even that is not ideal, as the module that I have loses it's configuration after power loss...

I've just downloaded the espresence project from github, and opened in vscode with platform.io, opened the `esp32s3-cdc` configuration in the platformio sidebar, clicked around to get the dependencies going and then done a `build and install` or something similar.

arbigator, 2025-01-10:
@vsviridov Thanks, I managed to get it installed, but I'm unable to access the /ui/#settings page. Tried on 3 different modules all with the same result. Seems to be the same problem as #1129, for which it seems no fix was found, just marked as stale and closed automatically.

I was actually using Bermuda prior, but wanted to check out ESPresense, though I guess I'll just stick with Bermuda until I get some different ESP devices to try out.

closed 2025-07-25 by stale[bot]