ESPresense/ESPresense #1298: Only one (first) ESPresense node registered on MQTT device list

softhuntermike, 2024-06-02:
### Describe the bug

After having the 1st node hooked up to MQTT server and appeared on MQTT device list, the 2nd one did not appear there, although the 2nd node did work.

andrewjallen16, 2024-10-18:
I'm having the same problem. Have you found any solutions?

dxmnkd316, 2024-10-18:
I think you're going to need to post a bit more info. What was the error?  What steps did you take to get there? What hardware are you using?  How did you install the firmware?

closed 2025-04-30 by stale[bot]