ESPresense/ESPresense #1303: Support for M5NanoC6

hajar97, 2024-06-10:
### Is your feature request related to a problem? Please describe.

Do you have any plans to add support for M5NanoC6 platform? This is the smallest form factor and more advanced than M5Atom, it has been around for a while now, but still no support

### Describe the solution you'd like

Firmware update that supports M5NanoC6

stefan-schweiger, 2024-06-13:
duplicate of https://github.com/ESPresense/ESPresense/issues/1270

DTTerastar, 2026-10-31:
ESP32-C6 support landed in #2181 and ships as `esp32c6` builds in the v5 betas ([v5.0.0b5](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b5)). That build should cover the M5NanoC6 as a generic C6 board, though it hasn't been tested on this board specifically. Please open a new issue if anything board-specific doesn't work.

closed 2026-10-31 by DTTerastar