ESPresense/ESPresense #1316: ESPresense relay output

scargill, 2024-06-30:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

You have PWM and RGB,RGB etc, how about s simple relay on-off? 

### Describe alternatives you've considered

I'm using the PWM but even at full brilliance one of my ESP32s is flickering no matter what power supply I use

DTTerastar, 2026-10-09:
Adding a concrete use case: **ESP32 smart plugs as nodes.** ESP32-C3 plugs like the Athom Smart Plug V3 (PG03V3-US16A — relay on GPIO5, LED on GPIO6, button on GPIO3) run ESPresense fine, but there's no way to drive the relay properly, so the outlet is either always off or wired up as a fake LED.

Today's workaround is to set the relay pin as an LED (`led_N_type` = PWM, `led_N_cntrl` = MQTT, `led_N_state` = on). That gets you an on/off light entity in Home Assistant, but:

- it shows up as a **light**, not a **switch**
- PWM on a relay pin is the wrong drive mode (and is presumably related to the flicker reported above)
- the power-on state handling is LED-oriented, not "what should this outlet do after a power cut"

What a proper relay output would want:

- [ ] Output type **Relay** (plain digital on/off), with an **inverted** option
- [ ] **Power-on state**: off / on / restore last
- [ ] Published to HA via MQTT discovery as a `switch`, with the usual `…/set` and state topics
- [ ] Optional: bind a button pin to toggle the relay locally, so the plug's own button still works

This is also a building block for per-board templates (#2529): a plug template needs a way to say "GPIO5 is the relay".

DTTerastar, 2026-10-31:
Shipped in #2530, available from the v5 betas ([v5.0.0b5](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b5)). The `outputs` hardware setting gives plain on/off relay outputs: `output` or `output_inverted` type, power-on state off/on/restore, a Home Assistant switch on `<room>/output_N` with `/set`, and an optional linked input so a button toggles the relay locally. The Athom plug example from this thread would be `[{"name": "Relay", "pin": 5, "type": "output", "power_on": "restore", "input": 1}]`.

closed 2026-10-31 by DTTerastar