ESPresense/ESPresense #1339: enroll apple watch irk

kei81131, 2024-08-02:
### Describe the bug

it is no an issue

Would like to share, I've enrolled apple watch irk using espresense instead of macos
Just like iphone, apple watch connected to espresense directly in setting > bluetooth page
Then, we can find the irk in espresense ui > devices page
Enjoy

raptile, 2024-08-17:
what do u mean?

devices > enroll?

Got two iPhones and both are pairing successfully according to the phone but no irk is visible on that page.
Worked fine with an iPad.

amaisano, 2024-08-23:
Are you saying you did not need to use the special Bluetooth LE Terminal WatchOS app? 

I just tried the special Watch app today and it didn't work. I had to use the Keychain lookup process.

AndLindemann, 2024-09-24:
You don't need the Bluetooth LE Terminal WatchOS app.
The native Bluetooth settings on the Watch would let me pair with Espresense (version 3.3.4) just fine. Just did it on an Apple Watch Ultra (gen 1) and a new Series 10.