ESPresense/ESPresense #1346: Cant find any settings to calibrate BME280 BMP280 and BH1750

kylevidrine, 2024-08-13:
### Describe the bug

I am recently getting into room presence detections and purchased a SEN0395, BME280, BMP280, and BH1750, then soldered all of them onto a board which I plan to put in a 3d printed case for several of my rooms. The only issue I have is that I can not find any place to calibrate the BME280 BMP280 and BH1750. The BME280 and BMP280 are reporting temps about 10 degrees higher than what the actual room is. After seeing several videos from Everything Smart Home on youtube I became interested in this project and would like to see if there is a way to calibrate the above sensors.

ChrWieg, 2024-12-30:
I have the very same issue, too high values (about 10°C) with 2 nodes with one BME280 at each node.

dxmnkd316, 2025-01-02:
I don't think there is a way unless you fork and write your own firmware build.  You'd need to change the calibration constants (BMP280_REGISTER_DIG_T1, T2, and T3) in a custom BMP280 library.  I don't know how you'd even determine what to change those constants to in order to actually calibrate them.  Or if you'd even want to do that.  

See these:
https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.cpp
https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.h

I would generally recommend just creating a helper and subtracting the delta from there.  If you've got a full 10 degC delta, either your sensor is damaged, the connection/soldering is bad, the case isn't breathing enough and transferring too much heat to the sensor, or the power supply is faulty.  

Some other devices I use from other IOT manufacturers report temperature but are all about 5-10 degF too high.  This is inevitably reporting ambient temp PLUS whatever heat is being generated from the electronics on the device.  This can be fairly substantial especially with an ESP32.

The power supply is extremely important for humidity calcs I've found.  If I have noisy voltage I have extremely noise humidity readings.  Less so for temperature, but it's still there.  High quality power supplies are very much recommended for temp/humidity measurement.  AliExpress power supplies have loads of issues in my experience when dealing with sensors.  That said, others have had great success with them.

One step you could take to determine if this is an ESPresense issue (unlikely) you could flash the ESP32 with ESPHome and see what the BMP280 is reporting there.  But again, I think this is extremely unlikely and either bad sensor, connection, or case are the most likely causes.

ChrWieg, 2025-01-03:
Thank you very much for this detailed answer! I will test the suggested points and post the results here.