ESPresense/ESPresense #1354: Seeed Studio XIAO ESP32C6

ingo916, 2024-08-22:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

Support for Seeed Studio XIAO ESP32C6

### Additional context

Is the esp32c3.bin  Seeed Studio XIAO ESP32C3 controller? If so what is the process for flashing the firmware?

bisquit2003, 2024-10-19:
+1
Please add support for seeed studio xiao esp32c6. They are ideal in size and stability of operation

ep4iot, 2025-06-24:
Please add support foe esp32c6 xiao. Would provide Wifi6, zigbee, matter compatibility for newer technology.

DTTerastar, 2026-10-31:
ESP32-C6 support landed in #2181 and ships as `esp32c6` builds in the v5 betas ([v5.0.0b5](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b5)). That build should cover the XIAO ESP32C6 as a generic C6 board, though it hasn't been tested on this board specifically. Please open a new issue if anything board-specific doesn't work.

closed 2026-10-31 by DTTerastar