ESPresense/ESPresense #1355: ESPHome loses all paired devices

kaciker, 2024-08-22:
### Describe the bug

For several days, ESPHome has been losing all paired devices., the MQTT room that is set by cfilecode is working, but the devices lose their assigned names after being paired with the ESPresence ESP32. If you check the devices, all previously paired devices are no longer paired. I have 5 ESP32 units, and this issue occurs with all of them. I suspect the problem might be related to MQTT, and it seems to happen after a Home Assistant reboot.  In the last 4 days I paired all phones again 4 times.

Is it possible to have a configuration file that maintains the relationship between the MAC address of each device and the friendly name assigned by ESPresence?

### Version

v3.3.2

### Active scan enabled

yes

### Include filter

no

### Exclude filter

yes

### Query filter

no

### Device (be specific)

ESP32

kaciker, 2024-08-22:
In the meantime, while this is being corrected, I found a way to be ready for the next time. I saved the MQTT devices' IRK configuration in a node inject in Node-RED. I tested this by deleting the topic in MQTT Explorer, and everything disappeared, but when I injected the saved data from the MQTT node, I was able to recover the configuration, and everything started working again

[image]

amaisano, 2024-08-23:
@kaciker could you please share your node red config/flow for this? Thanks.

kaciker, 2024-08-24:
> @kaciker could you please share your node red config/flow for this? Thanks.

Hello, The picture sent in the last msg was the only configuration needed. one inject node and one ouput node mqtt this last empty only with the connection mqtt, because the topic is inside the node inject.
You need to copy from Mqtt Explorer the topic and the payload, and write that in the flow

[image]
[image]

vhamburger, 2024-08-27:
All - NodeRed is nice - but if you do not have Node Red you can do it with HA just as easy (thanks a million for the tip @kaciker!! )
Here is a HA script that does the same - I tested it and it works like a charm. In case you lost the Matt entry - just run the script and you are back and running - super easy!

Here is the HA script: 
```
alias: Script - Restore ESpresence iPhone
sequence:
  - action: mqtt.publish
    metadata: {}
    data:
      qos: 0
      retain: false
      topic: espresense/settings/irk:<YOUR IRK HERE>/config
      payload: "{\"id\": \"<NAME>_iphone\", \"name\": \"<NAME> iPhone\"}"
description: ""
icon: mdi:restore
```
Enter your device name as `id` in lowercase as (no spaces - use "_" instead), `name` can be your device Name (should match what you had earlier in ESpresense HA config to "just work").

If the entry is missing again (MQTT reboot, whatever...) just run the script and you are back with a functioning device tracking ;) I have not found a way (yet) to automate, if anyone knows how to read MQTT via HA and trigger this as an automation that would be appreciated - but I hope this only needs to be a temporary workaround ;)

DTTerastar, 2025-02-08:
https://pagefault.blog/2020/02/05/how-to-set-up-persistent-storage-for-mosquitto-mqtt-broker/