ESPresense/ESPresense #1379: ESP32-C3 Enroll doesn't work

darkside9009, 2024-09-22:
### Describe the bug

Hi all together,

I got my C3 mini, I was able to flash online, connection is there Fingerprints are also there. But when I start the enroll function, I can see in my iPhone BL Espresense, I can connect it says it is connected but the the device window (with enroll) don't show, that it is connected, and the time still counting down till 0. In the fingerprints it is also missing.
I have tried 3.3.0 till 3.3.4 
I'm not able to see log in my serial connection.

### Version

3.3.4

dxmnkd316, 2024-09-23:
I've had this periodically.  Usually it's the iPhone not actually connecting properly.

Make sure you forget all ESPresense devices in the bluetooth settings.  Then I find that rebooting the iPhone and enrolling on that freshly booted device helps.  Make sure you're using the app recommended in the documentation.  Just connecting through the iPhone's bluetooth menu doesn't _always_ work.

darkside9009, 2024-10-14:
Hi,

I have tried to delete the the es presence from bluetooth and made a restart but it want to enroll.
My iPhone connect and show that its connected with es presence but the esp won't switch , it just counting till 0 and nothing happens.
I have tried with a ESP32 (not C3) here everything is working for few days, then the ESP loose the iBeacons and I have to enroll again, do I do something wrong?

dxmnkd316, 2024-10-14:
I've also had issues with Apple devices not playing well with the M5Stack Atom Lite S3.  I'm not sure about the C3 unfortunately.

Have you tried the app recommended by ESPresense? 
https://apps.apple.com/us/app/bluetooth-terminal/id1058693037

This worked best for me for both watches and iPhones when they were not connecting initially.

darkside9009, 2024-10-16:
I got the M5Stack Atom, and everything is working till now. I hope it won't loose the enrolled devices.
So there is definitely a bug in the C3 firmware.

cooljimy84, 2024-12-20:
I've got this on my esp32 s3 wroom 1. Trying to pair with an android phone, I'll see if I can flash th debug version and get a copy of the logs

erkr, 2025-02-03:
> I got the M5Stack Atom, and everything is working till now. I hope it won't loose the enrolled devices. So there is definitely a bug in the C3 firmware.

+1

cooljimy84, 2025-02-10:
Just tried 3.3.6b3 and its the same, just won't pair with my andriod, and nothing shows in the debug logs of the ESP device.

andresledo, 2025-02-28:
With my S25 ultra, even with my laptop I don't see the EspPresence device in the bluetooth options during the roll up, I tried all the available versions.

erkr, 2025-02-28:
> With my S25 ultra, even with my laptop I don't see the EspPresence device in the bluetooth options during the roll up, I tried all the available versions.

That is another issue. Devices show up, pairing starts but doesn't complete on the Espresense site

jmbruges, 2025-04-22:
> I got the M5Stack Atom, and everything is working till now. I hope it won't loose the enrolled devices. So there is definitely a bug in the C3 firmware.

I think the issue is the default autogenerated ID. I tried setting up manually the prefix in the ID, e.g., if I chose the phone then I keep the ID to the same name it generates without the "phone:". So far it worked in my iphone 14 after many attempts.

PR #1655 (2025-04-23): iOS 17/18 Enroll Fixed (fixes #1379, #1181)
fixes #1379
fixes #1181

## Summary by CodeRabbit

- **New Features**
  - Encrypted reads are now required for heart rate and device information characteristics, enhancing data security.

- **Bug Fixes**
  - Heart rate notifications are now throttled to a maximum of 4 times per second, ensuring consistent delivery.

- **Style**
  - Improved debug logging for notifications and status updates, providing clearer feedback during BLE operations.
commit 0bb38873 (2025-04-23): iOS 17/18 Enroll Fixed (fixes #1379, #1181) (#1655)

iOS 18 Enroll Fixed
closed 2025-04-23 by DTTerastar