ESPresense/ESPresense #1383: M5Atom-Lite LED not behaving correctly

AndLindemann, 2024-09-24:
### Describe the bug

I updated my M5Atom-Lites from 3.2.4 to 3.3.4 yesterday and noticed, that the LED is not working as it should.
LED 1 is configured with MQTT control.

[image]

But when I sent 
`{"state":"OFF"}`
the LED instead turns ON with maximum brightness.

Sending
`{"state":"ON","brightness":255}`
actually dims the LED a bit, it is definitely not as bright as before.

Sending
`{"state":"ON","brightness":10}`
seems to turn the LED OFF.

Sending
`{"state":"ON","brightness":3}`
turns the LED back ON with maximum brightness.

This has worked fine before on 3.2.4, so something must have changed. I tried the various different "LED type" settings, but none of them actually exhibits correct behavior.
Logs don't show anything specific that'd help with the issue.

### Version

3.3.4

### Device (be specific)

M5Stack Atom-Lite

ryanneufeld, 2024-11-25:
I am also experiencing this same bug. It's strange

stkang, 2024-12-12:
M5AtomS3, Works well with the settings below
PIN: 35
LED TYPE -> Addressable GRBW

InnovoDeveloper, 2025-02-24:
I am experience the same issue across several boards including esp32-C3 and esp32-S3 variants.  Seems to be across the board.  I am on the latest beta.

Kuifje88, 2025-03-18:
I have the same bug, with the following settings: A Neopixel ring connected to pin 33. Addressable GRB with 12 Count. I use MQTT for control. LED used to work with 3.3.2

closed 2025-09-28 by stale[bot]