ESPresense/ESPresense #1387: Wi-Fi time out

kendurbin, 2024-09-29:
### Describe the bug

when attempting to install program from Browser install, this fails to bring up active wi-fi connection found in it's query, which forces user to attempt to type in the desired wi-fi and credentials, only to find a time out attempting to connect.  
[image]

I dont have this issue when I attempt to load same board directly to ESPhome.  

suggestions?

### Version

3.3.4

Knobee, 2024-12-11:
Seems to be the same as #998 which has been open for a long time.

I'm having the same issue.

DTTerastar, 2025-02-08:
I have been unable to debug this so far.  It does kinda actually work even tho it does say timeout.  Try hitting terminal and see if it actually did configure the WiFi successfully.  Not sure why the web install doesn't understand what happened.