ESPresense/ESPresense #1403: Espresence loose all enrolled devices after few days

darkside9009, 2024-10-26:
### Describe the bug

Hello,
I have a problem, after few days all my enrolled devices are lost and I have to reenroll them again. 
I have tried with ESP32, C3 and ATOM Hardware, tried also with 3.3.4 and latest 3.3.5 software, all same.
I'm not sure if it's a bug or maybe a hidden setting, which I didn't found till now.
Please help, otherwise its not useable. 

### Version

3.3.5

mcsdodo, 2024-11-10:
Have multiple instances of ESP32 running the 3.3.5 and I'm not observing the issue. It would probably help if you posted your configuration.

slashinpdx, 2024-11-24:
I believe I read elsewhere that this is an issue of your MQTT Broker restarting and losing the enrolled devices.  Hope this helps.

DTTerastar, 2025-02-08:
Your MQTT Broker must support retain actually retaining things.  Check for an option to be persistent or retain.