ESPresense/ESPresense #1469: ERR_TOO_MANY_REDIRECTS when "Click here to edit other settings!"

jl-p, 2024-12-29:
### Describe the bug

I have 8 ESP PRESENSE ESP's in my house. and 3 i can still access but alle the others give the too many redirects messages when i want to go to the other settings page. the first page load without issues.  
I tried to reinstall one of them and choose erase all settings. i was able to access it once .  but now the issue is back
i deleted everything from my browser, cookies history extensions. but again the same.
i tried it from different devices with different browsers. but no luck.
any ideas. i normaly used the latest. so 3.3.5 and tried the latest beta . but no difference

### Version

3.3.5

### Active scan enabled

no

### Include filter

yes but want to add a device

### Exclude filter

no

### Firmware flavor

standard

### Device (be specific)

esp32 C3 super mini ,   and ESP32DEVKITV1

### Logs

```text
..................................192.168.1.8
Firmware:     esp32
Version:      v3.3.6b2
WiFi BSSID:   FC:EC:DA:3B:B3:B6 (channel=1 rssi=-46)
IP address:   192.168.1.8
DNS address:  192.168.1.1
Hostname:     espresense-woonkamer
Room:         Woonkamer
Mqtt server:  192.168.1.49:1883
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      iBeacon:***************** iBeacon:*********************
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 137636

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 Button | Button One: no Button Two: no
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-184-27445 to {"id":"node:badkamer","name":"Badkamer"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-56-51820 to {"id":"node:printkamer","name":"PrintKamer"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-32-14393 to {"id":"node:kantoor","name":"Kantoor"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-160-14140 to {"id":"node:meterkast","name":"Meterkast"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-148-63546 to {"id":"node:woonkamer","name":"Woonkamer"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-212-20619 to {"id":"node:slaapkamer","name":"Slaapkamer"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-232-12848 to {"id":"node:keuken","name":"Keuken"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-104-28034 to {"id":"node:badkamer","name":"Badkamer"}
1 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-148-63546 to node:woonkamer
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-148-63546 to {"id":"node:woonkamer","name":"Woonkamer"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-104-28034 to {"id":"node:badkamer","name":"Badkamer"}
```

### Screenshots

[image]

jpettitt, 2024-12-31:
+1 I'm also seeing this with 3.3.6b2

vezyr, 2025-01-18:
I believe the issue was introduced in the commit cf4f391.
A `transformIndexHtml` hook was added to the `cppPlugin`, which is supposed to handle all `.html` files and add them to `bundleAssets`. However, after checking the Vite documentation (https://vite.dev/guide/api-plugin#transformindexhtml), I noticed the following note:

> This hook won't be called if you are using a framework that has custom handling of entry files (for example SvelteKit).

It seems this is the case here. As a result, no `.html` files are added to `bundleAssets`, so no routes are generated for them.

vezyr, 2025-01-19:
Ok, after checking the commit history, I believe you can use version `v3.3.6b0` of ESPresense as a temporary solution. It should be the latest working version (with working 'other settings').

XandroTs, 2025-01-25:
I have the same problem too
Can you explain to me how I can go back to 3.3.6b0

Wheemer, 2025-01-25:
I did it by placing https://github.com/ESPresense/ESPresense/releases/download/v3.3.5b0/esp32.bin in the update path. Make sure it's the right one for your device.

DJ-CrossFade, 2025-01-25:
> I did it by placing https://github.com/ESPresense/ESPresense/releases/download/v3.3.5b0/esp32.bin in the update path. Make sure it's the right one for your device.

Do you mind telling me what the update path is? Where do I place this file?

Wheemer, 2025-01-25:
The main page of the web interface for your device, just above where you would have clicked to experience this issue in the first place. In the field marked "If set will update from this url on next boot:"

DJ-CrossFade, 2025-01-25:
> The main page of the web interface for your device, just above where you would have clicked to experience this issue in the first place. In the field marked "If set will update from this url on next boot:"

Oh sweet. Thanks. But check this out. I copied your link, pasted it to that on next boot box, saved at the bottom, then I assumed I needed to restart device at the top of the screen. Node disconnected then reconnected. I no longer get the too many redirects error but on the xxxx/ui/#settings page it's just blank.

[image]

XandroTs, 2025-01-26:
this is the right version and it works well for me

https://github.com/ESPresense/ESPresense/releases/download/v3.3.6b0/esp32.bin

Wheemer, 2025-01-26:
> > The main page of the web interface for your device, just above where you would have clicked to experience this issue in the first place. In the field marked "If set will update from this url on next boot:"
> 
> Oh sweet. Thanks. But check this out. I copied your link, pasted it to that on next boot box, saved at the bottom, then I assumed I needed to restart device at the top of the screen. Node disconnected then reconnected. I no longer get the too many redirects error but on the xxxx/ui/#settings page it's just blank.
> 
> [image]
> 
> Edited because I just confirmed it did roll back to your version.
> 
> [image]

I apologize, I linked the wrong version.

DJ-CrossFade, 2025-01-26:
Thanks for the correct version. I'm a give it a try tomorrow. Im pretty
sure it will fix my nodes. Really appreciate your help!

DoerakkerHA, 2025-02-03:
> this is the right version and it works well for me
> 
> https://github.com/ESPresense/ESPresense/releases/download/v3.3.6b0/esp32.bin

updating (up or downgrade) via the companion addon: 
[image]

[image]

DJ-CrossFade, 2025-02-03:
Crazy timing for you to say that. Took me a while to finally get around to moving them to this version. I literally got them running on that version like 2 hours ago. So far looks like it fixed the problem I was having as well.

DoerakkerHA, 2025-02-04:
> Crazy timing for you to say that. Took me a while to finally get around to moving them to this version. I literally got them running on that version like 2 hours ago. So far looks like it fixed the problem I was having as well. 

Curious: besides from being able to actually get to those pages ... what did you get from it (functionally)?

DTTerastar, 2025-02-08:
Fixed in v3.3.6b3

closed 2025-02-08 by DTTerastar

DoerakkerHA, 2025-02-09:
> Fixed in v3.3.6b3

Confirmed: off to update all my ESPrecense devices to V3.3.6b3

mhoogenbosch, 2025-02-09:
I thought I was up early, but an hour earlier (assuming GMT+1), but indeed. Just updated all 9 devices flawless. Thanks @DTTerastar !

DJ-CrossFade, 2025-02-10:
> Curious: besides from being able to actually get to those pages ... what did you get from it (functionally)?

On the extra page is where you change the maximum distance to report from 16 meters to 0 meters, which is needed in the ESPresense Companion add on. That's the only additional functionality I needed from that page but there are other features you can only get from that page too. Fingerprints probably being the most notable one.