ESPresense/ESPresense #1509: Can't open extra settings page

erkr, 2025-02-03:
### Describe the bug

I upgraded from v3.3.4 to v3.3.6b2 and cant open the extra settings page:

[image]

### Version

v3.3.6b2 

### Device (be specific)

ESP32c3

### Logs

```text
Can't open the serial port anymore since I flashed the first version. Upgrading via URL now
```

mhoogenbosch, 2025-02-08:
Duplicate of https://github.com/ESPresense/ESPresense/issues/1469

But solution is to install 3.3.6b0, is pretty much the same as 3.3.5 which is stable. The B1 introduces a nice feature to edit devices, but supposedly it also holds the bug.

Hopefully it'll get fixed soon, but last update is from December 11th.

DTTerastar, 2025-02-08:
Fixed in v3.3.6b3

closed 2025-02-08 by DTTerastar