ESPresense/ESPresense #1659: Stack overflow on ESP32-PoE when ethernet is connected

asolochek, 2025-04-28:
### Describe the bug

This may be a hardware problem, since this was working in the past. I noticed this device was offline, and when I reset it, and it came online and was providing valid info to MQTT, but I noticed from the LEDs it was still rebooting every 30 seconds or so. 

I reinstalled espresense on it via the browser, and selected 'erase device' and watched the console, and when I plugged it back into ethernet it started having stack overflows and resetting. When it's disconnected from ethernet and operating in AP mode, I can connect to it and it seems fine. 

### Version

3.3.5

### Firmware flavor

standard

### Device (be specific)

Olimex ESP32-PoE

### Logs

```text
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 215060
Setup network
Connecting to Ethernet. success!
Firmware:     esp32
Version:      v3.3.5
WiFi BSSID:    (channel=0 rssi=0)
IP address:   10.3.3.32
DNS address:  10.3.1.2
Hostname:     espresense-2c0b70
Room:         2c0b70
Mqtt server:  mqtt.z13.org:1883
Max Distance: 3.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      bacon carpaccio aarons_watch ham paio thaos_phone thaos_watch
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 138260

1 Motion | Pir: no Radar: no
1 Switch | Switch One: no Switch Two: no
1 Advert | iBeacon

***ERROR*** A stack overflow in task ble has been detected.

Backtrace: 0x40084041:0x3ffe2760 0x40096ad1:0x3ffe2780 0x400970c5:0x3ffe27a0 0x40097f41:0x3ffe2820 0x40096bcc:0x3ffe2850 0x40096b7e:0x0000a050 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

asolochek, 2025-05-06:
So, to rule out a hardware problem, I bought another ESP32-POE, but this time got the one with an external antenna. I found that the new device was also getting a stack overflow in the ble task, but it stopped if I removed the antenna. So it seems there is some device in the area that when seen by espresense on a ESP32-POE causes a stack overflow.

jb123567, 2025-06-22:
I have been having stack overflows for about 2 months connected to the Ethernet interface on 3 Olimex Devices running ESPHome, wonder if this is connected? Possibly the underlying code?