ESPresense/ESPresense #1676: enrolled ios device is lost after node restart

fedulovivan, 2025-05-05:
### Describe the bug

Steps:
- Use Iphone (15 pro with ios 17.1.1)
- Single esprsence node is running
- Enroll iphone in espresence UI, set device category "phone" and custom name "ivanf", check IRK is assigned
- Check messages could be received in specific topic. Eg **espresense/devices/phone:ivanf/room1**
- Restart esp32 device right from espresence UI (also reproducible with hardware button restart, and restart after cutting power off)

Expected:
- List of enrolled devices persisted, messages are coming to **espresense/devices/phone:ivanf/room1**

Actual:
- Previously added device is missing on Devices tab
- No messages in **espresense/devices/phone:ivan/room1** topic
- No IRK data on Devices tab

### Version

v3.3.5

### Device (be specific)

ESP32-D0WDQ6 (ESP32 NODEMCU 30pin DEVKIT V1)

### Logs

```text

```

hruba202, 2025-08-09:
i have 3 iphones(10,12,14),all 3 disappear and i have to enroll them again.again  and again ........

stulandr, 2026-03-11:
I've not looked at this since Oct last year and unfortunatey low priority for me atm.

closed 2026-10-09 by stale[bot]