ESPresense/ESPresense #1704: "Configurable WiFi Sleep Mode in Web UI"

BenAhrdt, 2025-05-30:
### Is your feature request related to a problem? Please describe.

Is it possible to implement an "on / off" configuration for:
the sleep mode of wifi (in esp32 default true.

" WiFi.setSleep(true/false) "

maxi1134, 2025-07-21:
Is this why my nodes keep switching to offline until I reboot them?!

DTTerastar, 2026-11-07:
Sorry for the bot noise earlier: #2098 was closed without merging. I'm closing this as not planned, and here's why.

We measured it on esp32, esp32-c3 and esp32-c6. Keeping the WiFi radio awake (sleep off) cut the BLE adverts each node saw by **33-41%**. The node shares one radio between WiFi and BLE, and WiFi modem sleep is what frees it to listen for BLE. A toggle would make it easy to give up a third of tracking data in exchange for a stability fix nobody has shown.

The real problem here was nodes dropping offline until rebooted. v5 addresses that differently. The WiFi reconnect loop was rewritten, and nodes now resend their Home Assistant discovery after every MQTT reconnect (#2598), so a node that comes back also shows up again in HA.

If you still see nodes going offline on [v5.0.0b8](https://github.com/ESPresense/ESPresense/releases/tag/v5.0.0b8), please open a new issue with the serial or TCP log from around the drop. That will tell us what's actually failing.

closed 2026-11-07 by DTTerastar